KlipperScreen/KlipperScreen #98: Bed_screws levelling issue

simonvez, 2021-02-25:
trying to understand the py script to manual bed levelling. I have no probe, all manual. I have a section defined as [bed_screws], but the klipperscreen will not go to where the screws are defined. It will offset them by 78 or somtething like that. when I use the klipper BED_SCREWS_ADJUST gcode command, it goes to where I define the [bed_screws]. It would be nice if the LCD would do the same.

jordanruthe, 2021-02-25:
Can you post your KlipperScreen.log file? The bed level panel takes into account probe offsets, but since you don't have a probe it should be going exactly to your screen locations.

simonvez, 2021-02-25:
hi Jordan. I dont know why but today it does go to the screw position. So im closing this one.

closed 2021-02-25 by simonvez