KlipperScreen/KlipperScreen #101: firmware retract

MSergW, 2021-02-28:
hi tell me how to add firmware retract setting to the print control menu?
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md#firmware-retraction

tonywright, 2022-04-08:
I would like to second this, it would be really useful to have firmware retraction available on the printer control panel. I have had a look at adding a panel and got as far as creating the UI but the values for the retraction are coming from the config file instead of the actual "Live" values as changed elsewhere. IE in printer.cfg I have firmware retraction length set to 0.5, if I adjust this to 0.75 in mainsail front-end  KlipperScreen still shows the 0.5 value not the adjusted value of 0.75 (Which doesn't get stored in config) This is different to acceleration which does work correctly.

alfrix, 2022-04-08:
> I would like to second this, it would be really useful to have firmware retraction available on the printer control panel. I have had a look at adding a panel and got as far as creating the UI but [...]

it's hard to say without looking or trying the code, you may just be missing the process_update in the panel to grab moonraker responses, you can take a look at how other panels do this

alfrix, 2022-04-08:
and to the OP, in you can sort of do this in KS but you need some advanced macros to go along with it, and it will be far worse than doing a dedicated panel
```
[menu __main config FWRT]
name: Firmware Retract
icon: retract

[menu __main FWRT config retract_length+]
name: Length +
icon: retract
gcode:
    MACRO_that_adds_more_retraction

[menu __main config FWRT retract_length-]
name: Length -
icon: retract
gcode:
    MACRO_that_substract_retraction

[menu __main config FWRT speed+]
name: Speed +
icon: speed+
gcode:
    MACRO_that_adds_more_speed

[menu __main config FWRT speed-]
name: Speed -
icon: speed-
gcode:
    MACRO_that_slows_retraction
```

tonywright, 2022-04-08:
alfrix, you were right (sort of) I had to add "firmware_retraction" to two places in "screen.py". my fwretraction.py panel is now working well.  How do I go about asking to add the two lines to screen.py? Otherwise it will stop working every time there is an update. I am happy to give you the panel file if you want to add it to the project for others including the OP

alfrix, 2022-04-10:
you can do a PR with the panel if you want

or attach it here, whatever you like

tonywright, 2022-04-11:
Although the panel works ok I can only make it available in KlipperScreen when the printer isn't printing. Is there a way to add it to the settings screen that is shown while printing?  It also requires the modification to screen.py to add the firmware retraction settings. I have attached it here (remove the .txt extension)
[fwretraction.py.txt](https://github.com/jordanruthe/KlipperScreen/files/8463688/fwretraction.py.txt)

alfrix, 2022-04-12:
i see that you copied from the limits panel, i had to do a lot of extra work, but i have it working now

alfrix, 2022-04-12:
and thank you tony i appreciate the help

commit 8ba28fde (2022-04-12): Add firmware retraction panel close #101
closed 2022-04-12 by alfrix