KlipperScreen/KlipperScreen #102: Z-Probe Calibration

nicedevil007, 2021-02-28:
If we do Z-Probe Calibration with KlipperScreen the calibration is started without checking if the printer is really homes. F.e. If you homed your printer in the past and then moved the printhead out of this position for anything else (filament change f.e.) the z-calibrate will start at this not centered state! That can damage the printbed while the probe is outside the bed during this movement.

Maybe it is possible to add a check if the printer is at its homed position or just let it home every time one is clicking on the probe calibration procedure.

robvoi, 2021-03-14:
I would add one more thing here. I have a detachable probe. If I forgot to put it on ....
So maybe a warning screen would be great. "did you home and have your probe ready?"

mstansberry, 2021-06-17:
Had a similar issue last night with [KlickyProbe](https://github.com/jlas1/Klicky-Probe/blob/main/Klipper%20macro/klicky-probe.cfg#L348) and the overrides for the `PROBE_CALIBRATE` macro where it attached the probe then tried to calibrate off the bed. Would it be simpler to just remove the automatic `PROBE_CALIBRATE` when entering the screen and leave it up to the user to make sure the probe is in a good position (similar to other manual `PROBE_CALIBRATE` runs)?

PR #327 (2021-11-21): Zcal panel: Support for Z_ENDSTOP_CALIBRATE
Also now it doesn't start until the button is pressed
i have left the abort button always active in case the calibration was started in the web interface, in that case you can just press start and continue with your calibration or abort.

Should Fix  #236 and also Fix #102
[image]
[image]
closed 2021-11-21 by jordanruthe