KlipperScreen/KlipperScreen #103: klipperScreen don't boot up

SanirS, 2021-03-02:
After finished all installation process,klipperScreen won't come up (last script line on screen "raspbain GNU/LINUX 10 mainsailos" "mainsailos login:")

log from: journalctl -xe -u KlipperScreen

pi@mainsailos:~/KlipperScreen $ journalctl -xe -u KlipperScreen
Mar 02 11:24:20 mainsailos systemd[1372]: KlipperScreen.service: Failed to execu
Mar 02 11:24:20 mainsailos systemd[1372]: KlipperScreen.service: Failed at step 
-- Subject: Process /usr/bin/xinit could not be executed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The process /usr/bin/xinit could not be executed and failed.
-- 
-- The error number returned by this process is ERRNO.
Mar 02 11:24:20 mainsailos systemd[1]: KlipperScreen.service: Main process exite
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit KlipperScreen.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 203.
Mar 02 11:24:20 mainsailos systemd[1]: KlipperScreen.service: Failed with result
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit KlipperScreen.service has entered the 'failed' state with result 'ex
lines 5854-5876/5876 (END)

Using pi 4 with 480X320 TFT LCD 
Thanks, 
Appreciate your work,

jordanruthe, 2021-03-04:
It looks like /usr/bin/xinit is not working properly. Can you verify that xinit exists?

SanirS, 2021-03-04:
thanks for the response,

After reinstalling the xinit and make sure it's working,  the KlipperScreen still won't come but know its seams to be for another reason, the only fail on the pi screen says "failed to start raise network interfaces"

 log: 
Mar 04 20:49:22 mainsailos xinit[676]: 2021-03-04 20:49:22,625 [menu.py:arrangeMenuItems()] - Evaluating item: preheat
Mar 04 20:49:22 mainsailos xinit[676]: DEBUG:root:Evaluating item: actions
Mar 04 20:49:22 mainsailos xinit[676]: 2021-03-04 20:49:22,625 [menu.py:arrangeMenuItems()] - Evaluating item: actions
Mar 04 20:49:22 mainsailos xinit[676]: DEBUG:root:Evaluating item: config
Mar 04 20:49:22 mainsailos xinit[676]: 2021-03-04 20:49:22,626 [menu.py:arrangeMenuItems()] - Evaluating item: config
Mar 04 20:49:22 mainsailos xinit[676]: DEBUG:root:Evaluating item: print
Mar 04 20:49:22 mainsailos xinit[676]: 2021-03-04 20:49:22,626 [menu.py:arrangeMenuItems()] - Evaluating item: print
Mar 04 20:49:22 mainsailos xinit[676]: DEBUG:root:Attaching panel main_panel
Mar 04 20:49:22 mainsailos xinit[676]: 2021-03-04 20:49:22,629 [screen.py:show_panel()] - Attaching panel main_panel
Mar 04 20:49:22 mainsailos xinit[676]: DEBUG:root:Current panel hierarchy: ['main_panel']
Mar 04 20:49:22 mainsailos xinit[676]: 2021-03-04 20:49:22,634 [screen.py:show_panel()] - Current panel hierarchy: ['main_panel']
Mar 04 20:49:22 mainsailos xinit[676]: DEBUG:root:Sending printer.objects.subscribe: {'objects': {'bed_mesh': ['profile_name', 'mesh_max', 'mesh_min', 'probed_matrix'], 'configfile': [
Mar 04 20:49:22 mainsailos xinit[676]: 2021-03-04 20:49:22,634 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe: {'objects': {'bed_mesh': ['profile_name',
Mar 04 20:49:41 mainsailos xinit[676]: DEBUG:root:Attempting to reconnect
Mar 04 20:49:41 mainsailos xinit[676]: 2021-03-04 20:49:41,937 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Mar 04 20:49:41 mainsailos xinit[676]: DEBUG:root:Reconnected
Mar 04 20:49:41 mainsailos xinit[676]: 2021-03-04 20:49:41,937 [KlippyWebsocket.py:reconnect()] - Reconnected
Mar 04 20:51:21 mainsailos sudo[851]: pam_unix(sudo:auth): conversation failed
Mar 04 20:51:21 mainsailos sudo[851]: pam_unix(sudo:auth): auth could not identify password for [pi]
Mar 04 20:53:21 mainsailos sudo[878]: pam_unix(sudo:auth): conversation failed

jordanruthe, 2021-03-04:
It looks like sudo is trying to authenticate for some commands that are used for wireless. Try running `./scripts/sudo_fix.sh` from the KlipperScreen folder

SanirS, 2021-03-04:
its giving me "Permission denied"