KlipperScreen/KlipperScreen #104: Bed screws adjust screen rotate

cinadr, 2021-03-02:
Hi!

Is it possible to rotate bed screws screen? It is rotated 180° compared to my real bed orientation.

Thank you.

apolkosnik, 2022-01-15:
Maybe reordering them in printer.cfg would do the trick?

cinadr, 2022-01-17:
Ok I looked up the code and found this logic:

Panel initialization:
1. Screws are _sorted_ by coordinates of Y then X position specified in printer.conf [screws_tilt_adjust] section] or [bed_screws] if exists. ([0,0] coords are at screws[0])
2. BLTouch offset added from [bltouch] section if it exists
3. The below table is calculated to assign links between screen coordinates (grid), buttons and screw coordinates.
    a. screws[2]: label[back left]    -> button[top left]         -> grid[2,0]
    b. screws[3]: label[back right] -> button[top right]       -> grid[1,0]
    c. _screws[0]_: label[front left]    -> _button[bottom left]_   -> grid[2,1]
    d. screws[1]: label[front right] -> button[bottom right] -> grid[1,1]
    e. grid 0,0 is Home All
    f. grid 0,1 is Disable XY
    g. grid 3,0 is Screws Adjust button

Panel Update processing:
1. Upon executing adjustment the result is identified by coordinates, the above _order is not changed_.

In my opinion the displayed bed orientation should depend on two factors. The **orientation of the bed** (and the coordinates of its origin) and the **orientation of the displaying screen** itself. For the user to be able to quickly find the proper screw to adjust and this screen to be useful both the bed and the display orientation **should match**. 

The above algorithm does not take into account neither of these parameters hence it is not displaying the bed picture in the correct orientation in some situations. The developer of this script falsely assume that the origin (0,0) point of the bed is the front left corner (bottom left button). The coordinate system is assumed to be positive in Y axis to to rear screws and the X axis pointing to the right.
The order of the screws defined in the printer.cfg file is irrelevant due to the **sorted** statement of the algorithm. 

I propose an option in KlipperScreen.cfg in a [bed_level] section to be able to assign screws to screen freely so the user can decide which orientation is useful to him. 
E.g.:
[bed_level]
frontleft:screw1   #bottomleft postion on screen
frontright:screw3 #bottomright on screen
rearleft:screw0     #topleft on screen
rearright:screw2  #topright on screen

Or something similar.

Thank you.
Zsolt.

tmsteinhardt, 2022-01-21:
I'm running into this same issue.  Has there been any update/solution or just need to live with the buttons being inverted for now?