KlipperScreen/KlipperScreen #108: Remaining print time is always completely of

sttts, 2021-03-10:
While printing the remaining time shown by Klipper screen does not match what Maintail shows. I tried with "file time" setting and "filament" setting, neither is near the Maintail values (which pretty much match the print).

Compare these photos, take within a minute:

Filament:
[image]

File:
[image]

Maintail:
[image]

jordanruthe, 2021-03-12:
I'll look into this, but it should be using the same logic

sttts, 2021-03-12:
Also note that the shown print is sometimes wrong, i.e. it shows an old preview picture. Maybe it mixes up old and new state?

sttts, 2021-03-13:
I think it is what I wrote in the previous comment, there is overlapping state between current and old print. Compare the following screenshot. It is while heating up for a new print, i.e. like a minute into the print, not 34min.

[image]

alexcroce, 2021-04-11:
Also for me, the remaining time is a lot more than what fluidd says.

matthewlloyd, 2021-05-19:
Another example, Fluidd is accurate here:

[image]
[image]

jordanruthe, 2021-05-19:
> Another example, Fluidd is accurate here:
> 
> [image]
> [image]

Can you attach you KlipperScreen.log file?

matthewlloyd, 2021-05-19:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6505187/KlipperScreen.log)

I have no KlipperScreen.conf so it's using defaults - file progress mode I think.

Roughly contemporaneous with the above photos:

```
2021-05-18 21:02:58,098 [screen.py:_websocket_callback()] - [
  "notify_gcode_response",
  "SD printing byte 12738412/17727772"
]
```

closed 2022-03-10 by alfrix

b14ckw1d0w, 2024-01-05:
This may be closed but the issue remains. I have the same issue with Mainsail currently showing 78% and KlipperScreen showing 67%. I have found another recent user on Reddit with the same issue. Something is not right somewhere with the logic.

alfrix, 2024-01-05:
Currently klipperscreen will not match the other interfaces since the percentage is calculated via estimated time, so totalPritnTime - remainingTime, it's not a bug it's intentional

alfrix, 2024-01-05:
Let's compare to current webui percentage:
90% of the web UI means 90 of the gcode

thearduinoguy, 2025-03-02:
As of 2nd March 2025 this issue persists.

0xUnkn0wN, 2025-03-04:
> Let's compare to current webui percentage:
> 
> 90% of the web UI means 90% of the gcode file done
> 
> If you completed 3 hours of a 4 hours print, would you say that is 90%?
> 
> Ks will show you 75%

That seems a bit awkward, time estimation is never 100 % correct and would fluctuate mid-print, so if i'm not wrong the percentage could go down as an example while printing because of the estimation goes up.
It would jump based on the current speed and acceleration as well.

With executed lines of gcode this could not happen, because we always know how many lines of gcode needs to be executed and how many already were executed.

PR #378 (2022-03-10): Rework Time estimates
It should be much more accurate
Fixes #222
Fixes #108 
Fixes #282