KlipperScreen/KlipperScreen #111: KlipperScreen exists from tracking print and goes back to main screen

michelluccote, 2021-03-18:
KlipperScreen exists from tracking print and goes back to main screen. I had a print run and when are returned after two hours KlipperScreen was no longer tracking the print and didn't realize it was still going.

I've attached the last many lines of log files to help.

[moonraker_tail.log](https://github.com/jordanruthe/KlipperScreen/files/6167121/moonraker_tail.log)
[klippy_tail.log](https://github.com/jordanruthe/KlipperScreen/files/6167405/klippy_tail.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6167406/KlipperScreen.log)

michelluccote, 2021-03-18:
I will add that when I do a KlipperScreen restart it does detect the job running and goes to the correct screen after.

jordanruthe, 2021-09-01:
Sorry for never commenting on this. I've made changes over time that will hopefully have solved this. Please reopen this if you're still having the issue.

closed 2021-09-01 by jordanruthe