KlipperScreen/KlipperScreen #112: Allow connecting to multiple printers.

al3ph, 2021-03-20:
I've only got one KlipperScreen, but two printers, would it be possible to support swapping the target printer, I assume you could just swap the host/port you connect to. (I assume it's moonraker you talk to).

cheers

Prutsium, 2021-03-20:
I have it with 2 printers ... seems to be implemented already.
Just define 2 printers

# Define printer and name. Name is anything after the first printer word
[printer Prusa Blue]
# Define the moonraker host/port if different from 127.0.0.1 and 7125
moonraker_host: 127.0.0.1
moonraker_port: 7125
# Moonraker API key if this is not connecting from a trusted client IP
moonraker_api_key: False

# Define printer and name. Name is anything after the first printer word
[printer Prusa Red]
# Define the moonraker host/port if different from 127.0.0.1 and 7125
moonraker_host: 127.0.0.1
moonraker_port: 7126
# Moonraker API key if this is not connecting from a trusted client IP
moonraker_api_key: False

al3ph, 2021-03-20:
Lol so it does. brilliant!

closed 2021-03-20 by al3ph