KlipperScreen/KlipperScreen #115: KlipperScreen install issue (vext)

pkmbau, 2021-03-23:
Hey guys,

I have an issue and no idea how to solve it.  I installed KlipperScreen first via Kiauh, after it doesnt work I removed it and installed it again via the official installation guide.
While installing the follow error appears:

**Requirement already satisfied: ruamel.yaml.clib>=0.1.2 in /home/pi/.KlipperScreen-env/lib/python3.7/site-packages (from ruamel.yaml>=0.                                                11.10->vext>=0.7.4->vext.gi==0.7.4) (0.2.2)
Skipping wheel build for vext.gi, due to binaries being disabled for it.
Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-ezr5                                                v8m_/vext-gi_0b940683d78746d28c284346ef4597de/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ezr5v8m_/vext-gi_0b940683d78746d28c284346ef                                                4597de/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();                                                exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-4ks70fpg/install-record.txt --single-version-externally                                                -managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-ezr5v8m_/vext-gi_0b940683d78746d28c284346ef4597de/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-4ks70fpg/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.arg                                                v[0] = '"'"'/tmp/pip-install-ezr5v8m_/vext-gi_0b940683d78746d28c284346ef4597de/setup.py'"'"'; __file__='"'"'/tmp/pip-install-ezr5v8m_/v                                                ext-gi_0b940683d78746d28c284346ef4597de/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r                                                \n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-4ks70fpg/install-recor                                                d.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check                                                 the logs for full command output.
Installing KlipperScreen unit file
Adding X11 Xwrapper
allowed_users=anybody
pi@raspberrypi:~/KlipperScreen $ cd**

So I checked the journalctl -xe -u KlipperScreen and got this message:
**-- Automatic restarting of the unit KlipperScreen.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Mar 23 16:20:40 raspberrypi systemd[1]: Stopped KlipperScreen.
-- Subject: A stop job for unit KlipperScreen.service has finished
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A stop job for unit KlipperScreen.service has finished.
--
-- The job identifier is 1793 and the job result is done.
Mar 23 16:20:40 raspberrypi systemd[1]: Started KlipperScreen.
-- Subject: A start job for unit KlipperScreen.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit KlipperScreen.service has finished successfully.
--
-- The job identifier is 1793.
Mar 23 16:20:41 raspberrypi xinit[1524]: X.Org X Server 1.20.4
Mar 23 16:20:41 raspberrypi xinit[1524]: X Protocol Version 11, Revision 0
Mar 23 16:20:41 raspberrypi xinit[1524]: Build Operating System: Linux 5.4.0-54-generic armv8l Raspbian
Mar 23 16:20:41 raspberrypi xinit[1524]: Current Operating System: Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l
Mar 23 16:20:41 raspberrypi xinit[1524]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1024 bcm27
Mar 23 16:20:41 raspberrypi xinit[1524]: Build Date: 15 January 2021  02:03:57PM
Mar 23 16:20:41 raspberrypi xinit[1524]: xorg-server 2:1.20.4-1+rpt2+deb10u3 (https://www.debian.org/support)
Mar 23 16:20:41 raspberrypi xinit[1524]: Current version of pixman: 0.36.0
Mar 23 16:20:41 raspberrypi xinit[1524]:         Before reporting problems, check http://wiki.x.org
Mar 23 16:20:41 raspberrypi xinit[1524]:         to make sure that you have the latest version.
Mar 23 16:20:41 raspberrypi xinit[1524]: Markers: (--) probed, (**) from config file, (==) default setting,
Mar 23 16:20:41 raspberrypi xinit[1524]:         (++) from command line, (!!) notice, (II) informational,
Mar 23 16:20:41 raspberrypi xinit[1524]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Mar 23 16:20:41 raspberrypi xinit[1524]: (==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Tue Mar 23 16:20:41 2021
Mar 23 16:20:41 raspberrypi xinit[1524]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Mar 23 16:20:41 raspberrypi xinit[1524]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Mar 23 16:20:41 raspberrypi xinit[1524]: (EE)
Mar 23 16:20:41 raspberrypi xinit[1524]: Fatal server error:
Mar 23 16:20:41 raspberrypi xinit[1524]: (EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
Mar 23 16:20:41 raspberrypi xinit[1524]: (EE)
Mar 23 16:20:41 raspberrypi xinit[1524]: Please consult the The X.Org Foundation support
Mar 23 16:20:41 raspberrypi xinit[1524]:          at http://wiki.x.org
Mar 23 16:20:41 raspberrypi xinit[1524]:  for help.
Mar 23 16:20:41 raspberrypi xinit[1524]: (EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
Mar 23 16:20:41 raspberrypi xinit[1524]: (EE)
Mar 23 16:20:41 raspberrypi xinit[1524]: (EE) Server terminated with error (1). Closing log file.**

Hopefully anybody has an idea how to solve this problem.

Thanks and best regards!

jordanruthe, 2021-03-26:
You can safely ignore the vext.gi issues. The real problem is permission denied on console two. Are you running another x windows such as the Linux desktop? Is the pi (or the username you are using) user in the tty group?

closed 2021-05-22 by jordanruthe

maruhe, 2021-05-25:
same here
it's just an octopi - no other x windows..

May 25 16:44:49 octopi xinit[796]: Build Operating System: Linux 5.4.0-54-generi
May 25 16:44:49 octopi xinit[796]: Current Operating System: Linux octopi 4.19.7
May 25 16:44:49 octopi xinit[796]: Kernel command line: coherent_pool=1M 8250.nr
May 25 16:44:49 octopi xinit[796]: Build Date: 19 April 2021  03:23:36PM
May 25 16:44:49 octopi xinit[796]: xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://
May 25 16:44:49 octopi xinit[796]: Current version of pixman: 0.36.0
May 25 16:44:49 octopi xinit[796]:         Before reporting problems, check http
May 25 16:44:49 octopi xinit[796]:         to make sure that you have the latest
May 25 16:44:49 octopi xinit[796]: Markers: (--) probed, (**) from config file,
May 25 16:44:49 octopi xinit[796]:         (++) from command line, (!!) notice,
May 25 16:44:49 octopi xinit[796]:         (WW) warning, (EE) error, (NI) not im
May 25 16:44:49 octopi xinit[796]: (==) Log file: "/home/pi/.local/share/xorg/Xo
May 25 16:44:49 octopi xinit[796]: (==) Using system config directory "/usr/shar
May 25 16:44:49 octopi xinit[796]: (EE)
May 25 16:44:49 octopi xinit[796]: Fatal server error:
May 25 16:44:49 octopi xinit[796]: (EE) xf86OpenConsole: Cannot open virtual con
May 25 16:44:49 octopi xinit[796]: (EE)
May 25 16:44:49 octopi xinit[796]: Please consult the The X.Org Foundation suppo
May 25 16:44:49 octopi xinit[796]:          at http://wiki.x.org
May 25 16:44:49 octopi xinit[796]:  for help.
May 25 16:44:49 octopi xinit[796]: (EE) Please also check the log file at "/home
May 25 16:44:49 octopi xinit[796]: (EE)
May 25 16:44:49 octopi xinit[796]: (EE) Server terminated with error (1). Closin
lines 804-826/826 (END)
May 25 16:44:49 octopi xinit[796]: Build Operating System: Linux 5.4.0-54-generic aarch64 Debian
May 25 16:44:49 octopi xinit[796]: Current Operating System: Linux octopi 4.19.71-rt24-v8+ #1 SMP PREEMPT RT Mon Mar 15 13:29:53
May 25 16:44:49 octopi xinit[796]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=256M video=HDMI-A-1:800x480M@59 vc_m
May 25 16:44:49 octopi xinit[796]: Build Date: 19 April 2021  03:23:36PM
May 25 16:44:49 octopi xinit[796]: xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support)
May 25 16:44:49 octopi xinit[796]: Current version of pixman: 0.36.0
May 25 16:44:49 octopi xinit[796]:         Before reporting problems, check http://wiki.x.org
May 25 16:44:49 octopi xinit[796]:         to make sure that you have the latest version.
May 25 16:44:49 octopi xinit[796]: Markers: (--) probed, (**) from config file, (==) default setting,
May 25 16:44:49 octopi xinit[796]:         (++) from command line, (!!) notice, (II) informational,
May 25 16:44:49 octopi xinit[796]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
May 25 16:44:49 octopi xinit[796]: (==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Tue May 25 16:44:49 2021
May 25 16:44:49 octopi xinit[796]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
May 25 16:44:49 octopi xinit[796]: (EE)
May 25 16:44:49 octopi xinit[796]: Fatal server error:
May 25 16:44:49 octopi xinit[796]: (EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
May 25 16:44:49 octopi xinit[796]: (EE)
May 25 16:44:49 octopi xinit[796]: Please consult the The X.Org Foundation support
May 25 16:44:49 octopi xinit[796]:          at http://wiki.x.org
May 25 16:44:49 octopi xinit[796]:  for help.
May 25 16:44:49 octopi xinit[796]: (EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional
May 25 16:44:49 octopi xinit[796]: (EE)
May 25 16:44:49 octopi xinit[796]: (EE) Server terminated with error (1). Closing log file.

pi@octopi:~ $ groups pi
pi : pi adm tty dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio
pi@octopi:~ $

my config.txt ends with
hdmi_force_edid_audio=1
max_usb_current=1
hdmi_force_hotplug=1
config_hdmi_boost=7
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_drive=1
hdmi_cvt 800 480 60 6 0 0 0

hsl416604093, 2021-10-18:
> same here it's just an octopi - no other x windows..
>
> May 25 16:44:49 octopi xinit[796]: Build Operating System: Linux 5.4.0-54-generi May 25 16:44:49 octopi xinit[796]: Current Operating System: [...]

HI，Has the problem been solved? Now i have the same problem！