KlipperScreen/KlipperScreen #117: Bed mesh is offset if done right after bed tilt screw calibration

michelluccote, 2021-03-26:
I did a few screws tilt runs to adjust my bed screws. I went straight to bed mesh calibration and realized it ends up with an offset. To fix this a home must be done between the two steps. Maybe this is not specific to KlipperScreen but maybe KlipperScreen can avoid the problem. If not done the bed mesh is offset which can result in either the nozzle digging into the bed or the being too far away from the bed.

danorder, 2021-04-03:
seems like a dying project. its klipper specific in docks they say you have to home it every time you run it.  once its level its good to home it before running a mesh. quite clunky

jordanruthe, 2021-12-04:
This is a late response, but you have to home Z between any Z calibration and doing anything with it. I'll consider if there are things I can do within KlipperScreen to help address, but it would be better to do this from klipper

closed 2021-12-04 by jordanruthe