KlipperScreen/KlipperScreen #122: KlipperScreen stays in the printed state after Canceled Print / Aborted Print

nicedevil007, 2021-04-05:
As the title says,

after a complete fresh boot on my pi I start a new print and this print was killed by an error (short2gnd f.e.) or cancled by myself.
The window goes back to "Restart | Main Menu" and after hitting the main menu button the screen of the Progress during the halted print is back again. Tested this multiple times now.

On Fluidd everything is ok, I can start a new print there without any problem.

Maybe the reset file isn't initiated correctly?

MSergW, 2021-04-19:
the same behavior when canceling printing and restarting the firmware - the screen is connected and shows the progress of the canceled printing, it is impossible to start a new print from the screen

DaStivi, 2021-04-19:
i've got this also once... even klipperscreen restart didn't helped... 🤔

csdstudio78, 2021-04-25:
Same... If I cancel a print then the print status on screen will not update again unless I reboot the PI.  During the next the print the screen functions as though there is nothing printing. I can see the bed/hot end temps.

jordanruthe, 2021-05-08:
Can anyone provide a log file?

fishboy7296, 2021-05-15:
This happened to me today.  Canceled print and print screen never properly updated after restarting print from fluidd (unable to restart from KlipperScreen).  Was able to fix by restarting KlipperScreen service via ssh (KlipperScreen  restart from system menu does not seem to work.). Attached is my log file. 

[https://gist.github.com/fishboy7296/e3b8ce00d9eb7675d33f3a8fb3b67d93]()

danorder, 2021-07-03:
same steps to repeat cancel print 2x boom screwed every time. to get around it restart klipper screen service same problem for monthsss

SaulGoodman1337, 2021-07-17:
same problem here

jordanruthe, 2021-07-18:
> same steps to repeat cancel print 2x boom screwed every time. to get around it restart klipper screen service same problem for monthsss

I'm unable to reproduce this. Can you tell me your exact steps? I am doing cancel print, restart print, cancel print, restart print, and everything is still working as expected.

lanman1, 2021-07-23:
I am getting this as well.  I was 1/2 way into 1st layer and hit cancel on the screen, it asked if I was sure and I said yes, then that dialog went away and it went back to a 'static'  print status screen.  I had to restart klipper and klipper firmware and then klipperscreen went back to the main screen.   During the 'lockup' I was able to go into the functions as if it was still printing.

jordanruthe, 2021-07-23:
Can you upload the klipperscreen log file?

lanman1, 2021-07-23:
Sorry to be dense, Where is it?  I don't see it in ~KlipperScreen

jordanruthe, 2021-07-23:
/tmp/KlipperScreen.log

lanman1, 2021-07-23:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6871268/KlipperScreen.log)

lanman1, 2021-07-23:
I'm trying to find the time it happened.  I think around 16:47 but I'm looking at the logs right now

Drachenkaetzchen, 2021-07-29:
Here's a partial log from KlipperScreen when it happened to me:

```
2021-07-29 20:28:43,617 [KlippyWebsocket.py:set_tool_temp()] - Sending set_tool_temp: M104 T0 S245.0
2021-07-29 20:28:59,702 [screen.py:_menu_go_back()] - #### Menu go back
2021-07-29 20:29:00,865 [screen.py:_menu_go_back()] - #### Menu go back
2021-07-29 20:29:02,447 [KlippyWebsocket.py:print_pause()] - Sending printer.print.pause
2021-07-29 20:29:03,290 [printer.py:change_state()] - Changing state from 'printing' to 'paused'
2021-07-29 20:29:03,291 [printer.py:change_state()] - Adding callback for state: paused
2021-07-29 20:29:03,540 [job_status.py:set_state()] - Changing job_status state from 'printing' to 'paused'
2021-07-29 20:29:04,774 [job_status.py:cancel_confirm()] - Canceling print
2021-07-29 20:29:04,776 [KlippyWebsocket.py:print_cancel()] - Sending printer.print.cancel
2021-07-29 20:29:09,288 [printer.py:change_state()] - Changing state from 'paused' to 'ready'
2021-07-29 20:29:09,289 [printer.py:change_state()] - Adding callback for state: ready
2021-07-29 20:29:14,871 [job_status.py:cancel_confirm()] - Canceling print
2021-07-29 20:29:14,873 [KlippyWebsocket.py:print_cancel()] - Sending printer.print.cancel
```

It seems that I accidentally hit pause, but I'm pretty sure this happens also without hitting pause.

Drachenkaetzchen, 2021-07-31:
When I cancel the print from KlipperScreen, and it stays in the job status screen as if it was still printing, I can do a "Clear File" from Fluidd and the job status screen changes to the options "Restart" and "Main Menu". 

I think KlipperScreen waits for the file to be cleared. Would be good if someone else can try this.

lanman1, 2021-07-31:
I too am running Fluidd but am on holiday so I can't test.  I will test
that as soon as I get back.

Drachenkaetzchen, 2021-08-20:
In the meanwhile I did other observations:

- If I `cancel` in KlipperScreen, it gets stuck until a "Clear File" (`SDCARD_RESET_FILE`) is issued. So this could be reproduced even if no web frontend is used.
- It feels like the first `cancel` in KlipperScreen behaves more like a pause.
- If I `pause` in KlipperScreen, then `cancel`, it works as expected.
- I guess that a missing `[virtual_sdcard]` in Klipper could be the reason it might not be reproducible by others

lanman1, 2021-09-23:
Mine did this and am still in this state now.  Is there any testing/info I can provide while it is in this state?  I cancelled from the screen and in Fluidd it did cancel but the screen is still showing it as an active print.  Temps are updating and I can go into settings but not back to the main menu.

danorder, 2021-09-23:
you have to reset klipper screen service from the panel to get it to sync
again. not sure what else is needed for steps to repeat other then cancel
print enough times in a row it will happen.

bluepinner, 2021-10-06:
I had this problem. It went away after adding SDCARD_RESET_FILE to the CANCEL_PRINT gcode macro.
I.E.

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
  SDCARD_RESET_FILE
  TURN_OFF_HEATERS
  CANCEL_PRINT_BASE

jordanruthe, 2021-12-04:
There's been updates to address this. If you are still having issues, please re-open this.

closed 2021-12-04 by jordanruthe