KlipperScreen/KlipperScreen #123: After update, Klipper Screen no longer starts

NovaViper, 2021-04-08:
I installled KlipperScreen via KIAUH, and it was just working until I accidentally ran an update on it and it updated KlipperScreen to the very latest commit that was done earlier. However now KlipperScreen just blanks, and then goes back to the terminal repeatedly.

I have an error like this in
```
Apr 08 06:32:05 printerpi xinit[2288]:     https://numpy.org/devdocs/user/troubleshooting-importerror.html
Apr 08 06:32:05 printerpi xinit[2288]: Please note and check the following:
Apr 08 06:32:05 printerpi xinit[2288]:   * The Python version is: Python3.7 from "/home/pi/.KlipperScreen-env/bin/python"
Apr 08 06:32:05 printerpi xinit[2288]:   * The NumPy version is: "1.20.2"
Apr 08 06:32:05 printerpi xinit[2288]: and make sure that they are the versions you expect.
Apr 08 06:32:05 printerpi xinit[2288]: Please carefully study the documentation linked above for further help.
Apr 08 06:32:05 printerpi xinit[2288]: Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory
Apr 08 06:32:05 printerpi xinit[2288]: /usr/bin/xinit: connection to X server lost
Apr 08 06:32:05 printerpi xinit[2288]: [1B blob data]
Apr 08 06:32:05 printerpi xinit[2288]: waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

```

henniero, 2021-04-08:
same issue for me. also trying to reinstall it completely doesnt't solve it. when the install script tries to update numpy i get a timeout warning.
I reverted back to the last comit with: `git reset --hard HEAD~1`
works fine again

NovaViper, 2021-04-08:
There's also this error when I go and install KlipperScreen

```
Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-mep4r8rs/vext-gi_b9510078a60949f1b94acae73c1c8ce3/setup.py'"'"'; __file__='"'"'/tmp/pip-install-mep4r8rs/vext-gi_b9510078a60949f1b94acae73c1c8ce3/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-v7q8_rvm/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi                                                            
         cwd: /tmp/pip-install-mep4r8rs/vext-gi_b9510078a60949f1b94acae73c1c8ce3/                                                          
    Complete output (22 lines):                                                                                                            
    running install                                                                                                                        
    vext.gi Install                                                                                                                        
    running build                                                                                                                          
    running install_data                                                                                                                   
    creating /home/pi/.KlipperScreen-env/share/vext                                                                                        
    creating /home/pi/.KlipperScreen-env/share/vext/specs                                                                                  
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs                                                                        
    running install_egg_info                                                                                                               
    running egg_info                                                                                                                       
    writing vext.gi.egg-info/PKG-INFO                                                                                                      
    writing dependency_links to vext.gi.egg-info/dependency_links.txt                                                                      
    writing requirements to vext.gi.egg-info/requires.txt                                                                                  
    writing top-level names to vext.gi.egg-info/top_level.txt                                                                              
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'                                                                                   
    reading manifest template 'MANIFEST.in'                                                                                                
    warning: no files found matching '*.txt'                                                                                               
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'                                                                                   
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info                       
    running install_scripts                                                                                                                
    writing list of installed files to '/tmp/pip-record-v7q8_rvm/install-record.txt'                                                       
    Install vext files:                                                                                                                    
    error: [Errno 2] No such file or directory: 'vext': 'vext'                                                                             
    ----------------------------------------                                                                                               
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-mep4r8rs/vext-gi_b9510078a60949f1b94acae73c1c8ce3/setup.py'"'"'; __file__='"'"'/tmp/pip-install-mep4r8rs/vext-gi_b9510078a60949f1b94acae73c1c8ce3/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-v7q8_rvm/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.
```

NovaViper, 2021-04-08:
@henniero I don't think I can go back to the old commit, since i had completely removed it off my system :(

henniero, 2021-04-08:
just reinstall (the actual broken version) and bump back with git reset afterwards. that worked for me.

henniero, 2021-04-08:
btw: had the same error at the end of the installation like you in vext.gi. just couldn't remeber the details because i didn't saved my terminal output.

NovaViper, 2021-04-08:
Oddly, I had to bump it back all to the commit before the two today, using `git reset --hard HEAD~2`, to the commit 791b1c3

henniero, 2021-04-08:
ooops. i think i actually also went back two commits. i put in the command twice.

phil65, 2021-04-08:
Installing `libatlas-base-dev` solved this for me.

Elektrickser, 2021-04-08:
Hi Jordan,

after install the update, i have a boot-loop of Klipper-Screen. Then i installed the Package "lib-atlas-base-dev" and i see Klipper-Screen but with-out connect to Printer. After Deinstallation and Re-Install, i had the follow Error in Kiuah, but the Klipper-Screen is here, but with-out Connect to printer.

`
Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jv42a2ur/vext-gi_e385aa59e0244e338b87ece1db77d604/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jv42a2ur/vext-gi_e385aa59e0244e338b87ece1db77d604/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-1kel2e36/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-jv42a2ur/vext-gi_e385aa59e0244e338b87ece1db77d604/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-1kel2e36/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jv42a2ur/vext-gi_e385aa59e0244e338b87ece1db77d604/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jv42a2ur/vext-gi_e385aa59e0244e338b87ece1db77d604/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-1kel2e36/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.
Installing KlipperScreen unit file
Adding X11 Xwrapper
allowed_users=anybody

>>>>>> KlipperScreen successfully installed!
`

Elektrickser, 2021-04-08:
Now i see, my mainsail is not working, too.

Elektrickser, 2021-04-08:
Now it works, after re-install Klipper and Moonraker.

jordanruthe, 2021-04-08:
All,

As @phil65 pointed out, I was missing a dependency I'm the requirements. I've pushed an update which will add that.

Depending on how you have it installed, you'll need to run the installation script again.

jordanruthe, 2021-04-08:
> There's also this error when I go and install KlipperScreen
>
> ```
> Installing collected packages: vext.gi
>     Running setup.py install for vext.gi ... error
>     ERROR: Command errored out with exit [...]

You can ignore that vext.gi error safely

NovaViper, 2021-04-08:
Installing `libatlas-base-dev` solved this for me! This package needs to be added as a dependency to avoid breakage

jordanruthe, 2021-04-08:
It has since been added as a dependency.

AndreasCr, 2021-04-09:
Version v0.1.4-73 doesn't work, either ... how to fix it ?

jordanruthe, 2021-04-09:
Either re-run the installation script or run `sudo apt-get install
libatlas-base-dev`

AndreasCr, 2021-04-09:
@jordanruthe Thank you, that works ... now the GUI starts up, again ... !

jordanruthe, 2021-04-19:
I'm going to close this as I also have updated the install script so that moonraker will install dependencies on updates going forward.

closed 2021-04-19 by jordanruthe