KlipperScreen/KlipperScreen #124: Prusa Thumbnails

alexcroce, 2021-04-10:
I cannot view job thumbnails on Klipperscreen, even if i can in fluidd. Prusa settings are G-code Thumbnails: 16x16, 
[image]
On klipperscreen i only view the .stl placeholder.

kylix34, 2021-04-10:
Use these dimensions for the thumbnails:
32x32,400x300

alexcroce, 2021-04-10:
> Use these dimensions for the thumbnails:
> 32x32,400x300

[image]

Same issue

jordanruthe, 2021-04-10:
Can you upload your log file after this error occurs? `/tmp/KlipperScreen.log`

alexcroce, 2021-04-10:
> Can you upload your log file after this error occurs? `/tmp/KlipperScreen.log`

2021-04-10 19:29:49,860 [printer.py:change_state()] - Changing state from 'ready' to 'printing'
2021-04-10 19:29:49,861 [printer.py:change_state()] - Adding callback for state: printing
2021-04-10 19:29:49,863 [screen.py:_load_panel()] - Loading panel: job_status
2021-04-10 19:29:49,864 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/job_status.py
2021-04-10 19:29:49,879 [screen_panel.py:__init__()] - Button scale: [1.2, 1.4]
2021-04-10 19:29:50,013 [screen.py:show_panel()] - Attaching panel job_status
2021-04-10 19:29:50,021 [job_status.py:update_file_metadata()] - Cannot find file metadata. Listening for updated metadata
2021-04-10 19:29:50,021 [job_status.py:activate()] - Act State: printing
2021-04-10 19:29:50,025 [screen.py:show_panel()] - Current panel hierarchy: ['job_status']
2021-04-10 19:44:29,771 [screen.py:_websocket_callback()] - [
  "notify_gcode_response",
  "Done printing file"
]

jordanruthe, 2021-04-10:
Ideally it would be the whole log file. Metadata is initially fetched near
startup.

alexcroce, 2021-04-10:
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/6290636/klippy.log)

alexcroce, 2021-04-10:
after a reboot it started to work.

closed 2021-04-19 by jordanruthe