KlipperScreen/KlipperScreen #125: Job status panel not showing gcode preview

kylix34, 2021-04-10:
Hi,

i installed MainsailOS with Klipper and I noticed the Job Status panel is not showing the gcode 3D model preview. It shows instead an .STL file icon.

kylix34, 2021-04-10:
Just noticed there's another opened issue with the same problem

closed 2021-04-10 by kylix34

TehProgrammer, 2021-12-10:
Please link the issue.  I'm also having this problem.

Tahutipai, 2021-12-11:
Same here.  

How does KlipperScreen find the STL file?

Tahutipai, 2021-12-11:
For anyone else looking...

In order to view the ".stl" on the screen, a thumbnail must be generated by your [slicer.
](https://docs.fluidd.xyz/features/thumbnails)

Deluxa, 2022-12-02:
maybe it will help someone else, I was solving the same problem.....
For Cura..
1. install plugin for Cura https://github.com/emtrax-ltd/Cura2MoonrakerPlugin
2. configure your Moonraker 3D printer in Cura..... Settings -> Printer -> Manage Printers -> Connect Moonraker
3. 
[image]
4.slice and select "upload to (your printer)"