KlipperScreen/KlipperScreen #131: Bed Level moves out of range

DaStivi, 2021-04-15:
Hi, when using the bed level in klipperscreen it looks like its not using the bed_screws positions from printer.cfg... somehow it moves out of range for the far left 2... and moves only about half of the bed for the right screws...

using the probe screw ajustment does work as intended... 
using the Bed Screw adjust from fluidd its also working as intened and the nozzle gets to the [bed_screws] position..
i even found another bug report #98 that magically solved itself :(

`Apr 15 22:39:15 octopi xinit[521]: DEBUG:root:Attaching panel main_panel_config_bed_level
Apr 15 22:39:15 octopi xinit[521]: ### Creating panel bed_level : bed_level {'name': "{{ gettext('Bed Level') }}", 'icon': 'bed-level', 'panel': 'bed_level', 'method': False, 'confirm': False, 'enable': True, 'params': {}}
Apr 15 22:39:15 octopi xinit[521]: 2021-04-15 22:39:15,911 [screen.py:show_panel()] - Attaching panel main_panel_config_bed_level
Apr 15 22:39:15 octopi xinit[521]: DEBUG:root:Current panel hierarchy: ['main_panel', 'main_panel_config', 'main_panel_config_bed_level']
Apr 15 22:39:15 octopi xinit[521]: 2021-04-15 22:39:15,938 [screen.py:show_panel()] - Current panel hierarchy: ['main_panel', 'main_panel_config', 'main_panel_config_bed_level']
Apr 15 22:39:17 octopi xinit[521]: DEBUG:root:Going to position: [-2.0, 29.0]
Apr 15 22:39:17 octopi xinit[521]: DEBUG:root:Sending printer.gcode.script: G90
Apr 15 22:39:17 octopi xinit[521]: G1 Z7 F800
Apr 15 22:39:17 octopi xinit[521]: G1 X-2.0 Y29.0 F3600
Apr 15 22:39:17 octopi xinit[521]: G1 Z.1 F300
Apr 15 22:39:17 octopi xinit[521]: 2021-04-15 22:39:17,048 [bed_level.py:go_to_position()] - Going to position: [-2.0, 29.0]
Apr 15 22:39:17 octopi xinit[521]: 2021-04-15 22:39:17,049 [KlippyWebsocket.py:gcode_script()] - Sending printer.gcode.script: G90
Apr 15 22:39:17 octopi xinit[521]: G1 Z7 F800
Apr 15 22:39:17 octopi xinit[521]: G1 X-2.0 Y29.0 F3600
Apr 15 22:39:17 octopi xinit[521]: G1 Z.1 F300
Apr 15 22:39:17 octopi python[520]: [websockets.py:build_error()] - JSON-RPC Request Error: 400
Apr 15 22:39:17 octopi python[520]: {'message': 'Move out of range: -2.000 29.000 6.785 [0.000]', 'error': 'WebRequestError'}
Apr 15 22:39:17 octopi python[520]: Traceback (most recent call last):
Apr 15 22:39:17 octopi python[520]:   File "/home/pi/moonraker/moonraker/websockets.py", line 136, in execute_method
Apr 15 22:39:17 octopi python[520]:     result = await method(ws, *args, **kwargs)
Apr 15 22:39:17 octopi python[520]:   File "/home/pi/moonraker/moonraker/websockets.py", line 205, in func
Apr 15 22:39:17 octopi python[520]:     WebRequest(endpoint, kwargs, conn=ws))
Apr 15 22:39:17 octopi python[520]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 419, in make_request
Apr 15 22:39:17 octopi python[520]:     return await self._request_standard(web_request)
Apr 15 22:39:17 octopi python[520]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 473, in _request_standard
Apr 15 22:39:17 octopi python[520]:     return await base_request.wait()
Apr 15 22:39:17 octopi python[520]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 619, in wait
Apr 15 22:39:17 octopi python[520]:     raise self.response
Apr 15 22:39:17 octopi python[520]: utils.ServerError: {'message': 'Move out of range: -2.000 29.000 6.785 [0.000]', 'error': 'WebRequestError'}
Apr 15 22:39:17 octopi xinit[521]: DEBUG:root:[
Apr 15 22:39:17 octopi xinit[521]:   "notify_gcode_response",
Apr 15 22:39:17 octopi xinit[521]:   "!! Move out of range: -2.000 29.000 6.785 [0.000]"
Apr 15 22:39:17 octopi xinit[521]: ]
Apr 15 22:39:17 octopi xinit[521]: 2021-04-15 22:39:17,927 [screen.py:_websocket_callback()] - [
Apr 15 22:39:17 octopi xinit[521]:   "notify_gcode_response",
Apr 15 22:39:17 octopi xinit[521]:   "!! Move out of range: -2.000 29.000 6.785 [0.000]"
Apr 15 22:39:17 octopi xinit[521]: ]
`

jordanruthe, 2021-04-16:
Can you post the full KlipperScreen.log file?

jordanruthe, 2021-06-12:
Please provide a log file if you are still having issues.

closed 2021-06-12 by jordanruthe