KlipperScreen/KlipperScreen #133: Changing the font size

Shadow2091, 2021-04-16:
Hey! Could you add the option to select the font size in the settings, since the standard font on a 4" screen seems very small to me, although the tiles and button icons are of a good size. Thank you!

jordanruthe, 2021-04-19:
Unfortunately, changing the font size to be bigger currently has some cascading problems with making items appear off the screen. 

I do want to fix those problems at some point to make the font bigger.

alfrix, 2021-05-04:
i did experiment with this some label cutting needs to happen because using Pango word wrap adds extra blank space at he bottom and it's problematic in some panels

This is [Big icons and text ](https://github.com/alfrix/KlipperScreen/commit/93398dcaf497835f021af70dce6a568172d87328)using the #144 as base:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

also made a font selector test
https://github.com/alfrix/KlipperScreen/commit/28d9ab7e3e775e81418673af6c8e3d34b7c78a55

SaulGoodman1337, 2021-07-17:
> 
> 
> i did experiment with this some label cutting needs to happen because using Pango word wrap adds extra blank space at he bottom and it's problematic in some panels
> 
> This is [Big icons and text ](https://github.com/alfrix/KlipperScreen/commit/93398dcaf497835f021af70dce6a568172d87328)using the #144 as base:
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> [image]
> 
> also made a font selector test
> [alfrix@28d9ab7](https://github.com/alfrix/KlipperScreen/commit/28d9ab7e3e775e81418673af6c8e3d34b7c78a55)

Hi,
thanks for this! but how can i use this?
i have a very cheap and simple 3.5 TFT and the fontsize here is very tiny.... 
i need a quickfix before i switch to a hyperpixel or something like that

alfrix, 2021-07-17:
Navigate the files on your pi and find
_/home/pi/KlipperScreen/ks_includes/KlippyGtk.py_
change
`font_ratio = [51, 30]`
to
`font_ratio = [43, 29]`

this will make the font 11px instead of 9px, (you can  do the other changes in #169)
currently it's the maximum you can have without issues

PR #245 (2021-09-11): Simple font size selector
Fix #133 
The difference is quite noticeable in all tested resolutions, here some samples from the two extremes

at 480x320 it ranges from 9px to 12px
[image]
[image]

and at 1080p ranges from 34px small  to 40px large
[image]
[image]
closed 2021-09-11 by jordanruthe