KlipperScreen/KlipperScreen #134: Change Panel infos

alexcroce, 2021-04-16:
Is it possibile to configure infos shown on a panel? I.e. if i want to show layers remaining in jobs panel, is it possible?

jordanruthe, 2021-04-19:
Currently, it's not possible. But this brings up a larger discussion I was looking at having. 

I'm not happy with all the information on the jobs panel and am looking to redesign it a bit. For instance, I don't believe the X/Y positions provide much value and will likely not display them going forward. 

I'm open to suggestions here, and I may try to figure a way to have selectable info for this panel, but I'm unsure of what that'd look like right now.

mpalm0508, 2021-04-23:
I agree with X/Y position.  I would like to see M117 messages if possible.  I have my slicers adding the current layer to a M117 message and it shows great in Fluidd.  Current speed might be good as well.

ohjay93, 2021-04-27:
I second the idea of M117 messages, that makes it easy for me as an end-user to change what the screen shows.
Other input: X/Y values are usefull when I am setting up home position, working on probe positions etc.
Layer height is reeeeeally useful for watching for filament change positions as well as for continueing failed prints :-)

zztopper, 2021-05-10:
Just made a PR #150 that adds a line to JobStatus panel with M117 contents, if you're still interested @ohjay93 @mpalm0508

alexcroce, 2021-05-11:
It looks great! Thank you!

closed 2021-12-04 by jordanruthe