KlipperScreen/KlipperScreen #137: Illegal instruction (core dumped)

kacpergogol-htd, 2021-04-21:
Hi,
I'm trying to port this code to docker and make it more portable over different devices then RPI.

But even on bare-metal i'm facing such issue:

```
DISPLAY=:0.0 ${KSENV}/bin/python /root/KlipperScreen/screen.py
Illegal instruction (core dumped)
```

I tryed to play with GDB but still nothing , have You got any ideas ?

Kind Regards,
Kacper

kacpergogol-htd, 2021-04-21:
Getting closer...
```
/root/.KlipperScreen-env/bin/python3 -c "import websocket"
Illegal instruction (core dumped)
```

kacpergogol-htd, 2021-04-22:
```
INFO:root:KlipperScreen version: v0.1.4-76-gbb86891
INFO:root:Passed config file: /root/KlipperScreen.conf
2021-04-22 00:08:11,842 [screen.py:main()] - KlipperScreen version: v0.1.4-76-gbb86891
2021-04-22 00:08:11,845 [config.py:get_config_file_location()] - Passed config file: /root/KlipperScreen.conf
INFO:root:Found configuration file at: /root/KlipperScreen/ks_includes/KlipperScreen.conf
DEBUG:root:Config path location: /root/KlipperScreen/ks_includes/KlipperScreen.conf
2021-04-22 00:08:11,846 [config.py:get_config_file_location()] - Found configuration file at: /root/KlipperScreen/ks_includes/KlipperScreen.conf
2021-04-22 00:08:11,846 [config.py:__init__()] - Config path location: /root/KlipperScreen/ks_includes/KlipperScreen.conf
DEBUG:root:Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2021-04-22 00:08:11,857 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
DEBUG:root:Setting up wifi event loop
2021-04-22 00:08:11,861 [wifi.py:run()] - Setting up wifi event loop
DEBUG:root:OS Language: C
2021-04-22 00:08:11,870 [screen.py:__init__()] - OS Language: C
/root/KlipperScreen/screen.py:106: DeprecationWarning: Gdk.Screen.get_width is deprecated
  self.width = self._config.get_main_config().getint("width", Gdk.Screen.get_width(Gdk.Screen.get_default()))
/root/KlipperScreen/screen.py:107: DeprecationWarning: Gdk.Screen.get_height is deprecated
  self.height = self._config.get_main_config().getint("height", Gdk.Screen.get_height(Gdk.Screen.get_default()))
INFO:root:Screen resolution: 800x480
DEBUG:root:img width: 50 height: 51
2021-04-22 00:08:11,997 [screen.py:__init__()] - Screen resolution: 800x480
2021-04-22 00:08:11,998 [KlippyGtk.py:__init__()] - img width: 50 height: 51
DEBUG:root:Loading panel: splash_screen
INFO:root:Panel path: /root/KlipperScreen/panels/splash_screen.py
2021-04-22 00:08:12,010 [screen.py:_load_panel()] - Loading panel: splash_screen
2021-04-22 00:08:12,011 [screen.py:_load_panel()] - Panel path: /root/KlipperScreen/panels/splash_screen.py
DEBUG:root:Button scale: [1.2, 1.4]
2021-04-22 00:08:12,036 [screen_panel.py:__init__()] - Button scale: [1.2, 1.4]
DEBUG:root:Attaching panel splash_screen
2021-04-22 00:08:12,124 [screen.py:show_panel()] - Attaching panel splash_screen
DEBUG:root:Current panel hierarchy: ['splash_screen']
2021-04-22 00:08:12,127 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
DEBUG:root:Power devices: []
2021-04-22 00:08:12,150 [splash_screen.py:show_restart_buttons()] - Power devices: []
DEBUG:root:Changing power save to: 3600
2021-04-22 00:08:12,173 [screen.py:set_screenblanking_timeout()] - Changing power save to: 3600
DEBUG:root:Printers: [{'Printer': {'moonraker_host': '127.0.0.1', 'moonraker_port': '7125', 'moonraker_api_key': ''}}]
2021-04-22 00:08:12,204 [screen.py:__init__()] - Printers: [{'Printer': {'moonraker_host': '127.0.0.1', 'moonraker_port': '7125', 'moonraker_api_key': ''}}]
INFO:root:Connecting to printer: Printer
2021-04-22 00:08:12,205 [screen.py:connect_printer()] - Connecting to printer: Printer
DEBUG:root:Button scale: [1.2, 1.4]
2021-04-22 00:08:12,209 [screen_panel.py:__init__()] - Button scale: [1.2, 1.4]
DEBUG:root:Attaching panel splash_screen
2021-04-22 00:08:12,238 [screen.py:show_panel()] - Attaching panel splash_screen
DEBUG:root:Current panel hierarchy: ['splash_screen']
2021-04-22 00:08:12,241 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
DEBUG:root:Power devices: []
2021-04-22 00:08:12,272 [splash_screen.py:show_restart_buttons()] - Power devices: []
DEBUG:root:Sending request to http://127.0.0.1:7125/machine/device_power/devices
2021-04-22 00:08:12,277 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-22 00:08:12,292 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
DEBUG:urllib3.connectionpool:http://127.0.0.1:7125 "GET /machine/device_power/devices HTTP/1.1" 404 317
[... 47 lines cut ...]

  File "/root/KlipperScreen/ks_includes/printer.py", line 103, in process_update
    self.evaluate_state()

  File "/root/KlipperScreen/ks_includes/printer.py", line 112, in evaluate_state
    idle_state = self.data['idle_timeout']['state'].lower() # possible values: Idle, printing, ready
NoneType: None
2021-04-22 00:08:13,006 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'state'
Traceback:   File "/root/KlipperScreen/screen.py", line 661, in init_printer
    self.printer.reinit(printer_info['result'], data)

  File "/root/KlipperScreen/ks_includes/printer.py", line 71, in reinit
    self.process_update(data)

  File "/root/KlipperScreen/ks_includes/printer.py", line 103, in process_update
    self.evaluate_state()

  File "/root/KlipperScreen/ks_includes/printer.py", line 112, in evaluate_state
    idle_state = self.data['idle_timeout']['state'].lower() # possible values: Idle, printing, ready
NoneType: None
DEBUG:root:Attempting to reconnect
DEBUG:root:Reconnected
2021-04-22 00:08:13,311 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-04-22 00:08:13,312 [KlippyWebsocket.py:reconnect()] - Reconnected
```

closed 2021-04-22 by kacpergogol-htd

gucio1200, 2021-05-06:
Just in case, I did not merge, couze it is WIP and MVP.
Idea can be sourced here:
- https://github.com/gucio1200/klipper-docker/