KlipperScreen/KlipperScreen #138: Bed Level Tool Displaying One Corner

rtran1995, 2021-04-23:
Bed leveling tool used to show all 3 corners and how much to rotate by, now I'm only able to see the back right after the most recent update. 
[image]
[image]

rtran1995, 2021-04-25:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6371490/KlipperScreen.log)

Bed leveling starts at `2021-04-25 02:38:17,188 [screen.py:_load_panel()] - Loading panel: bed_level`

jordanruthe, 2021-04-25:
I was not accounting for a situation of screws with a negative position value. I just pushed a change which should address and fix your issue.

rtran1995, 2021-04-26:
[image]
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6379741/KlipperScreen.log)
Fixed the bottom right and back right but back left still seems to be having issues.

Screw positions remain the same.

jordanruthe, 2021-04-27:
I missed a regex statement, let me know if this fixes the last screw for you.

rtran1995, 2021-04-27:
Confirmed fixes issues. Thank you very much!

closed 2021-04-27 by rtran1995