KlipperScreen/KlipperScreen #141: Add "Wi-Fi Config" button.

denziko76, 2021-04-26:
Good day! 
Please add a very useful "Wi-Fi Config" button.
So that you can scan and connect to any network without any problems. Imagine how convenient it will be when you move the printer to another place to connect to the network simply through the screen. No need to disassemble the printer or whatever.
Thank you for understanding.

jordanruthe, 2021-04-26:
Hey, I am currently working on this feature.

jordanruthe, 2021-05-14:
This has now been implemented

closed 2021-05-14 by jordanruthe

Tux0Topo, 2023-04-26:
Hi there! Sorry to necropost, but ive tried with Raspberry OS x86 and it only shows the current wifi connection. In the desktop, all of them are shown. Any ideas?
It was all installed by kiauh. Thanks!