KlipperScreen/KlipperScreen #143: Unable to load panel job_status

4noxx, 2021-04-29:
can any explain this message?

```
--------------------KlipperScreen Log Start--------------------
Git Version: v0.1.4-80-gb39d433
2021-04-29 07:45:05,895 [screen.py:main()] - KlipperScreen version: v0.1.4-80-gb39d433
2021-04-29 07:45:05,899 [config.py:get_config_file_location()] - Passed config file: /home/pi/KlipperScreen.conf
2021-04-29 07:45:05,901 [config.py:get_config_file_location()] - Found configuration file at: /home/pi/KlipperScreen/ks_includes/KlipperScreen.conf
2021-04-29 07:45:05,902 [config.py:__init__()] - Config path location: /home/pi/KlipperScreen/ks_includes/KlipperScreen.conf
2021-04-29 07:45:05,915 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2021-04-29 07:45:05,966 [wifi.py:run()] - Setting up wifi event loop
2021-04-29 07:45:05,966 [screen.py:__init__()] - OS Language: en_GB.UTF-8
2021-04-29 07:45:06,253 [screen.py:__init__()] - Screen resolution: 800x480
2021-04-29 07:45:06,254 [KlippyGtk.py:__init__()] - img width: 50 height: 51
2021-04-29 07:45:06,271 [screen.py:_load_panel()] - Loading panel: splash_screen
2021-04-29 07:45:06,272 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/splash_screen.py
2021-04-29 07:45:06,307 [screen_panel.py:__init__()] - Button scale: [1.2, 1.4]
2021-04-29 07:45:06,431 [screen.py:show_panel()] - Attaching panel splash_screen
2021-04-29 07:45:06,433 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2021-04-29 07:45:06,459 [splash_screen.py:show_restart_buttons()] - Power devices: []
2021-04-29 07:45:06,516 [screen.py:set_screenblanking_timeout()] - Changing power save to: 3600
2021-04-29 07:45:06,587 [screen.py:__init__()] - Printers: [{'Printer': {'moonraker_host': '127.0.0.1', 'moonraker_port': '7125', 'moonraker_api_key': ''}}]
2021-04-29 07:45:06,588 [screen.py:connect_printer()] - Connecting to printer: Printer
2021-04-29 07:45:06,591 [screen_panel.py:__init__()] - Button scale: [1.2, 1.4]
2021-04-29 07:45:06,622 [screen.py:show_panel()] - Attaching panel splash_screen
2021-04-29 07:45:06,624 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2021-04-29 07:45:06,649 [splash_screen.py:show_restart_buttons()] - Power devices: []
2021-04-29 07:45:06,654 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2021-04-29 07:45:06,663 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 07:45:06,673 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /machine/device_power/devices HTTP/1.1" 404 390
2021-04-29 07:45:06,678 [screen.py:connect_printer()] - Found power devices: False
2021-04-29 07:45:06,680 [files.py:__init__()] - 
2021-04-29 07:45:06,682 [screen.py:connect_printer()] - Connected to printer: Printer
2021-04-29 07:45:06,683 [selector_events.py:__init__()] - Using selector: EpollSelector
2021-04-29 07:45:07,181 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-04-29 07:45:07,182 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2021-04-29 07:45:07,191 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 07:45:07,206 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 200 46
2021-04-29 07:45:07,232 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2021-04-29 07:45:07,233 [KlippyWebsocket.py:on_open()] - Self.connected = False
2021-04-29 07:45:07,235 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2021-04-29 07:45:07,244 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 07:45:07,260 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/info HTTP/1.1" 200 553
2021-04-29 07:45:07,264 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&idle_timeout&configfile&gcode_move&fan&toolhead&virtual_sdcard&print_stats&heater_bed&extruder&pause_resume&webhooks
2021-04-29 07:45:07,274 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 07:45:07,487 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/objects/query?bed_mesh&idle_timeout&configfile&gcode_move&fan&toolhead&virtual_sdcard&print_stats&heater_bed&extruder&pause_resume&webhooks HTTP/1.1" 200 4058
2021-04-29 07:45:07,493 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2021-04-29 07:45:07,503 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 07:45:07,516 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /machine/device_power/devices HTTP/1.1" 404 390
2021-04-29 07:45:07,522 [printer.py:reinit()] - Moonraker object status: {'virtual_sdcard': {'progress': 0.0, 'is_active': False, 'file_position': 0}, 'print_stats': {'print_duration': 0.0, 'total_duration': 0.0, 'filament_used': 0.0, 'filename': '', 'state': 'standby', 'message': ''}, 'bed_mesh': {}, 'pause_resume': {'is_paused': False}, 'gcode_move': {'homing_origin': [0.0, 0.0, 0.0, 0.0], 'speed_factor': 1.0, 'gcode_position': [0.0, 0.0, 0.0, 0.0], 'absolute_extrude': True, 'absolute_coordinates': True, 'position': [0.0, 0.0, 0.0, 0.0], 'speed': 1500.0, 'extrude_factor': 1.0}, 'idle_timeout': {'state': 'Idle', 'printing_time': 0.0}, 'configfile': {'settings': {'virtual_sdcard': {'path': '~/gcode_files'}, 'printer': {'square_corner_velocity': 5.0, 'move_flush_time': 0.05, 'max_accel': 1000.0, 'buffer_time_start': 0.25, 'buffer_time_low': 1.0, 'max_velocity': 1000.0, 'kinematics': 'none', 'buffer_time_high': 2.0, 'max_accel_to_decel': 500.0}, 'pause_resume': {'recover_velocity': 50.0}, 'gcode_macro resume': {'gcode': '\n{% set E = printer["gcode_macro PAUSE"].extrude|float %}\nG91\nG1 E{E} F2100\nRESTORE_GCODE_STATE NAME=PAUSE_state\nBASE_RESUME', 'rename_existing': 'BASE_RESUME'}, 'gcode_macro pause': {'gcode': '\n{% set E = printer["gcode_macro PAUSE"].extrude|float %}\n{% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}\n{% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}\n{% set max_z = printer.toolhead.axis_maximum.z|float %}\n{% set act_z = printer.toolhead.position.z|float %}\n{% if act_z < (max_z - 2.0) %}\n{% set z_safe = 2.0 %}\n{% else %}\n{% set z_safe = max_z - act_z %}\n{% endif %}\nSAVE_GCODE_STATE NAME=PAUSE_state\nBASE_PAUSE\nG91\nG1 E-{E} F2100\nG1 Z{z_safe} F900\nG90\nG1 X{x_park} Y{y_park} F6000', 'variable_extrude': '1.0', 'rename_existing': 'BASE_PAUSE'}, 'gcode_macro cancel_print': {'gcode': '\nTURN_OFF_HEATERS\nCLEAR_PAUSE\nSDCARD_RESET_FILE\nBASE_CANCEL_PRINT', 'rename_existing': 'BASE_CANCEL_PRINT'}, 'idle_timeout': {'gcode': "\n{% if 'heaters' in printer %}\n   TURN_OFF_HEATERS\n{% endif %}\nM84\n", 'timeout': 600.0}, 'mcu': {'baud': 250000, 'serial': '/dev/serial/by-id/<your-mcu-id>', 'max_stepper_error': 2.5e-05}}, 'config': {'virtual_sdcard': {'path': '~/gcode_files'}, 'printer': {'max_velocity': '1000', 'kinematics': 'none', 'max_accel': '1000'}, 'pause_resume': {}, 'display_status': {}, 'gcode_macro RESUME': {'rename_existing': 'BASE_RESUME', 'gcode': '\n{% set E = printer["gcode_macro PAUSE"].extrude|float %}\nG91\nG1 E{E} F2100\nRESTORE_GCODE_STATE NAME=PAUSE_state\nBASE_RESUME'}, 'gcode_macro PAUSE': {'rename_existing': 'BASE_PAUSE', 'variable_extrude': '1.0', 'gcode': '\n{% set E = printer["gcode_macro PAUSE"].extrude|float %}\n{% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}\n{% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}\n{% set max_z = printer.toolhead.axis_maximum.z|float %}\n{% set act_z = printer.toolhead.position.z|float %}\n{% if act_z < (max_z - 2.0) %}\n{% set z_safe = 2.0 %}\n{% else %}\n{% set z_safe = max_z - act_z %}\n{% endif %}\nSAVE_GCODE_STATE NAME=PAUSE_state\nBASE_PAUSE\nG91\nG1 E-{E} F2100\nG1 Z{z_safe} F900\nG90\nG1 X{x_park} Y{y_park} F6000'}, 'gcode_macro CANCEL_PRINT': {'rename_existing': 'BASE_CANCEL_PRINT', 'gcode': '\nTURN_OFF_HEATERS\nCLEAR_PAUSE\nSDCARD_RESET_FILE\nBASE_CANCEL_PRINT'}, 'mcu': {'serial': '/dev/serial/by-id/<your-mcu-id>'}}, 'save_config_pending': False}, 'toolhead': {'square_corner_velocity': 5.0, 'max_accel': 1000.0, 'homed_axes': '', 'estimated_print_time': 0.0, 'max_velocity': 1000.0,  [... cut ...]
    temp_grid.attach(self.labels[self.current_extruder + '_box'], 0, 0, 1, 1)
KeyError: '_box'
2021-04-29 08:29:48,050 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel job_status
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 282, in show_panel
    self.panels[panel_name].initialize(panel_name)
  File "/home/pi/KlipperScreen/panels/job_status.py", line 95, in initialize
    temp_grid.attach(self.labels[self.current_extruder + '_box'], 0, 0, 1, 1)
KeyError: '_box'

4noxx, 2021-04-29:
after restart: see only KLIPPER - CONNECTING TO PRINTER screen
with RESTART - FIRMWARE RESTART - MENU (bottom right)

new LOG:
```
--------------------KlipperScreen Log Start--------------------
Git Version: v0.1.4-80-gb39d433
2021-04-29 08:40:33,199 [screen.py:main()] - KlipperScreen version: v0.1.4-80-gb39d433
2021-04-29 08:40:33,208 [config.py:get_config_file_location()] - Passed config file: /home/pi/KlipperScreen.conf
2021-04-29 08:40:33,211 [config.py:get_config_file_location()] - Found configuration file at: /home/pi/KlipperScreen/ks_includes/KlipperScreen.conf
2021-04-29 08:40:33,213 [config.py:__init__()] - Config path location: /home/pi/KlipperScreen/ks_includes/KlipperScreen.conf
2021-04-29 08:40:33,349 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2021-04-29 08:40:33,541 [wifi.py:run()] - Setting up wifi event loop
2021-04-29 08:40:33,563 [screen.py:__init__()] - OS Language: en_GB.UTF-8
2021-04-29 08:40:34,027 [screen.py:__init__()] - Screen resolution: 800x480
2021-04-29 08:40:34,028 [KlippyGtk.py:__init__()] - img width: 50 height: 51
2021-04-29 08:40:34,072 [screen.py:_load_panel()] - Loading panel: splash_screen
2021-04-29 08:40:34,073 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/splash_screen.py
2021-04-29 08:40:34,122 [screen_panel.py:__init__()] - Button scale: [1.2, 1.4]
2021-04-29 08:40:34,488 [screen.py:show_panel()] - Attaching panel splash_screen
2021-04-29 08:40:34,492 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2021-04-29 08:40:34,537 [splash_screen.py:show_restart_buttons()] - Power devices: []
2021-04-29 08:40:34,611 [screen.py:set_screenblanking_timeout()] - Changing power save to: 3600
2021-04-29 08:40:34,703 [screen.py:__init__()] - Printers: [{'Printer': {'moonraker_host': '127.0.0.1', 'moonraker_port': '7125', 'moonraker_api_key': ''}}]
2021-04-29 08:40:34,704 [screen.py:connect_printer()] - Connecting to printer: Printer
2021-04-29 08:40:34,709 [screen_panel.py:__init__()] - Button scale: [1.2, 1.4]
2021-04-29 08:40:34,750 [screen.py:show_panel()] - Attaching panel splash_screen
2021-04-29 08:40:34,756 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2021-04-29 08:40:34,790 [splash_screen.py:show_restart_buttons()] - Power devices: []
2021-04-29 08:40:34,797 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2021-04-29 08:40:34,819 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 08:40:34,877 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /machine/device_power/devices HTTP/1.1" 404 390
2021-04-29 08:40:34,883 [screen.py:connect_printer()] - Found power devices: False
2021-04-29 08:40:34,886 [files.py:__init__()] - 
2021-04-29 08:40:34,889 [screen.py:connect_printer()] - Connected to printer: Printer
2021-04-29 08:40:34,889 [selector_events.py:__init__()] - Using selector: EpollSelector
2021-04-29 08:40:35,872 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-04-29 08:40:35,873 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2021-04-29 08:40:35,881 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 08:40:35,893 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 200 46
2021-04-29 08:40:35,912 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2021-04-29 08:40:35,912 [KlippyWebsocket.py:on_open()] - Self.connected = False
2021-04-29 08:40:35,915 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2021-04-29 08:40:35,921 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 08:40:35,935 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/info HTTP/1.1" 404 390
2021-04-29 08:40:35,938 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/objects/query?bed_mesh&idle_timeout&configfile&gcode_move&fan&toolhead&virtual_sdcard&print_stats&heater_bed&extruder&pause_resume&webhooks
2021-04-29 08:40:35,949 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 08:40:35,960 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /printer/objects/query?bed_mesh&idle_timeout&configfile&gcode_move&fan&toolhead&virtual_sdcard&print_stats&heater_bed&extruder&pause_resume&webhooks HTTP/1.1" 404 390
2021-04-29 08:40:35,965 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2021-04-29 08:40:35,973 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
2021-04-29 08:40:35,986 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /machine/device_power/devices HTTP/1.1" 404 390
2021-04-29 08:40:35,992 [functions.py:logging_exception_handler()] - Uncaught exception <class 'TypeError'>: 'bool' object is not subscriptable
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 658, in init_printer
    data = data['result']['status']
NoneType: None
2021-04-29 08:40:36,370 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
2021-04-29 08:40:36,370 [KlippyWebsocket.py:reconnect()] - Reconnected
2021-04-29 08:40:37,650 [screen.py:_websocket_callback()] - [
[... cut ...]
2021-04-29 08:44:36,057 [screen.py:_websocket_callback()] - [
  "notify_gcode_response",
  "// Mesh Bed Leveling Complete"
]
2021-04-29 08:44:36,060 [screen.py:_websocket_callback()] - [
  "notify_gcode_response",
  "// Bed Mesh state has been saved to profile [default]\n// for the current session.  The SAVE_CONFIG command will\n// update the printer config file and restart the printer."
]
2021-04-29 08:44:44,659 [screen.py:_websocket_callback()] - [
  "notify_gcode_response",
  "// Unknown command:\"M1001\""
]

jordanruthe, 2021-04-29:
I can push a fix for this in a few days, in the meantime you can add
`[idle_timeout]` to your klipper configuration to fix this.

closed 2021-05-09 by jordanruthe

tripslip, 2022-07-31:
+1 I have the same issue. 
404 GET /machine/device_power/devices (127.0.0.1) [No User]

commit f8ca339d (2021-05-09): printer: Fix NoneType issue

Fixes #143