KlipperScreen/KlipperScreen #154: Thumbnails of new files not available

kmarty, 2021-05-12:
Hi,
Since commit 96d252ba3cb99a06a55d9588e38a4457783c0488 any uploaded file for print which doesn't exists previously has no thumbnail when printing starts.
After searching why, it seems files are scanned only when KlipperScreen starts (in _init_printer()_)

This "hotfix" works for me (I don't say this is the right way - I don't simply know):
```diff
diff --git a/screen.py b/screen.py
index 0e49d44..8a1e116 100644
--- a/screen.py
+++ b/screen.py
@@ -689,6 +689,7 @@ class KlipperScreen(Gtk.Window):
             del self.panels["job_status"]
 
     def printer_printing(self):
+        self.files.refresh_files()
         self.close_popup_message()
         self.show_panel('job_status',"job_status", "Print Status", 2)
 
```

jordanruthe, 2021-05-15:
Appreciate the feedback. This will be a temporary problem. Moonraker will shortly be implementing inotify which will remedy this problem. I may push something in the meantime depending on how long that will take

jordanruthe, 2021-05-15:
Here's the pr: https://github.com/Arksine/moonraker/pull/159

kmarty, 2021-05-15:
Oh, I see. ok.

jordanruthe, 2021-05-17:
Arksine is planning on merging the mentioned PR Tuesday around 7:30pm EDT (UTC-23:00)

jordanruthe, 2021-05-22:
This should be fixed. Please let me know if you're still having issues.

closed 2021-05-22 by jordanruthe

kmarty, 2021-05-23:
Seems it works.