KlipperScreen/KlipperScreen #155: Temperature issue

rjmessing, 2021-05-14:
I just installed Klipperscreen using Kiauh on a fresh install of RaspiOS, along with Fluidd, Klipper, and Moonraker.  I'm doing a PID calibration and was looking for the temperature on the screen when I noticed it wasn't there.  I went to controls and temperature and clicked on it.  Got an error and said check klipperscreen.log

UnboundLocalError: local variable 'primary_tool' referenced before assignment
2021-05-13 20:28:01,746 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel temperature
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 287, in show_panel
    self.panels[panel_name].initialize(panel_name)
  File "/home/pi/KlipperScreen/panels/temperature.py", line 38, in initialize
    self.labels[primary_tool].get_style_context().add_class('button_active')
UnboundLocalError: local variable 'primary_tool' referenced before assignment

jordanruthe, 2021-05-14:
Can you post the entire KlipperScreen.log file?

rjmessing, 2021-05-14:
Here is the log
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6476197/KlipperScreen.log)

jordanruthe, 2021-05-14:
I'm not seeing an extruder defined, but maybe it's something to do with klipper startup? If you pull the latest update and try again, let me know if that fixes it

rjmessing, 2021-05-15:
I don't know what i did to make it work.  My printer has been down due to a crash and subsequent troubleshooting to see if anything else broke, so it may have been just a issue as not homing my printer or with me messing with hotend stuff that ended up fixing the issue.  I also updated, but that was after I noticed the screen had temperature readings on it.  If it happens again I'll repost.

closed 2021-05-22 by jordanruthe