KlipperScreen/KlipperScreen #161: Restart KlipperScreen button on System panel not working

fishboy7296, 2021-05-19:
I was unable to get the "Restart KlipperScreen" button on the System panel to do anything.  I am assuming that the sudo command is/was looking for a password. Now there might be something wrong with my installation/configuration, but I was able to get it to work by configuring the command to run without a password by performing the following steps:

Execute the following command:
`sudo nano /etc/sudoers.d/KlipperScreen-restart`

Add the following text, where username is your username, and save the file:
`username ALL=NOPASSWD: /bin/systemctl restart KlipperScreen`

Not sure if this is something that can be added to the installation, or if there is a better way to do this, but thought I would share.

jordanruthe, 2021-05-19:
The installation script doesn't check for sudo access, currently. 

From the scripts folder, you need to run `sudo ./sudo_fix.sh`

fishboy7296, 2021-05-19:
Okay thanks, I forgot about needing the leading "./".  I finally got it to run, but first I had to make the script runnable by executing `chmod uga+x sudo_fix.sh`.  The "Restart Klipper Screen" button is now working without my suduers.d file I created earlier.

Thanks!

closed 2021-05-20 by jordanruthe