KlipperScreen/KlipperScreen #170: Gcode preheat function and custom preheat presets

davide-molino, 2021-05-28:
Hi, I have two suggestion: the first one is custom preheat setting, if I go to the config and change the value or name it returns to default everytime I update and th moonraker updater says that the installation is invalid. The second suggestion is to add a preheat function relative to gcode in the print menu, just like fluidd does.

jordanruthe, 2021-05-28:
You should be making config modifications to a separate file, ~/KlipperScreen.conf. That will solve issue number 1

davide-molino, 2021-05-28:
I already make modification to that file, but fluidd updater say that is invalid. and after any update the file return to the defaults

jordanruthe, 2021-05-28:
You shouldn't be modifying a file... If you're making changes to ks_includes/KlipperScreen.conf, it will complain about it being invalid. You have to make a new file in your home directory with that name

davide-molino, 2021-05-30:
thank you very much. Now it works... but think about a gcode preheat please... it will be very useful

closed 2021-12-04 by jordanruthe