KlipperScreen/KlipperScreen #171: Show additional pressure, humidity & gas reading from BMP680 sensor

henkyprayoga, 2021-05-29:
Hi,

Is it possible to show the current value of BMP680 sensor on the panel? The temperature works already. However BMP680 also has pressure, humidity and gas reading.

Thank you.

emersonsc, 2022-01-06:
I was hoping for the same. I really want to use this or the 280 for monitoring the Humidity and temps in my printer room, and want it pooled up the same as the temps for the MCU and RPI.

@henkyprayoga , can you provide the code you used to get the temp data to show up?

wiggywoosaa, 2023-10-17:
+1 for this feature

thijstriemstra, 2024-06-30:
+1

I've installed a BME280 and was expecting to see the humidity in klipperscreen like fluidd does:

[image]

ploucandco, 2024-08-06:
+1 for this feature