KlipperScreen/KlipperScreen #173: Klipperscreen won't connect to Klipper

seacred, 2021-05-31:
Hi there,
After the last Update to the newest Klipper-version klipperscreen seems like not connecting anymore.
Somebody else got this issue?
Thanks in advance!
Best regards

Potplanty, 2021-06-06:
Yep, on two machines.

jordanruthe, 2021-06-06:
Can you guys upload a log file?

Potplanty, 2021-06-06:
> Can you guys upload a log file?

Sure thing. Before I saw this I performed a full removal of the KlipperScreen folder and cloned it again from the Github repo. Ran the installer which went through without issues. Did a full reboot of the Pi after install and it now shows another message about missing resources. Log file attached as well.
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6604360/KlipperScreen.log)

jordanruthe, 2021-06-07:
@Potplanty In your case, an image was renamed and wasn't able to be found. I just pushed an update that will gracefully fail/continue in this case.

jordanruthe, 2021-06-07:
@alfrix You may want to update the icon on the solarized image.

Potplanty, 2021-06-07:
> 
> 
> @Potplanty In your case, an image was renamed and wasn't able to be found. I just pushed an update that will gracefully fail/continue in this case.

Thanks Jordan. Working fine now.

jordanruthe, 2021-06-11:
@seacred Are you still having this issue?

seacred, 2021-06-12:
Works great again.
Thanks for that.
Issue can be closed.

closed 2021-06-12 by seacred