KlipperScreen/KlipperScreen #182: KlipperScreen don't connect to Wi-Fi

AndrezinhoCosta, 2021-06-08:
Hi... 
5 ou 6 times I have installed the Octoprint with VCoreOS and when install KlipperScreen with PiTFT50 and reboot th Raspberry, I loose the wifi connection. I don't know what to do!

jordanruthe, 2021-06-08:
How are you setting up the wireless connection? are you changing wifi settings within KlipperScreen? It's hard for me to debug this without the KlipperScreen.log file, but I understand you can't necessarily get it without a wifi connection. Do you have the ability to plug the pi in via an ethernet cable after the error occurs so you can grab the log file?

AndrezinhoCosta, 2021-06-08:
I'm modifying the vcoreos-wpa-supplicant.txt with my Wi-Fi network and password normally. I have 4 Rasps working here with Octoprint and Logi C920e webcams.
I'm working to put this V-Core 3 assembling in my Youtube channel. But I'll close the printer and want to put the PiTFT50 on the front door with KlipperScreen.
I got a ethernet cable here! I'll try to see if I get connect to see the KlipperScreen.log.
In the boot screen it shows "Disabling power management for wlan0". I don't know if it helps...

André Luiz L. Costa
Facebook: Andrezinho Costa<https://www.facebook.com/Andrezinho.costahttp://>
Celular / WhatsApp: (021) 9-7117-1735
Banda Alma de Vinyl

AndrezinhoCosta, 2021-06-08:
I just made a fresh resinstall of VCoreOS here **WITHOUT** the KlipperScreen and got the same problem. 
So, it's not a KlipperScreen problem. 
I'll look for another solution... or get crazy trying! :)

jordanruthe, 2021-06-11:
Good luck!

closed 2021-06-11 by jordanruthe