KlipperScreen/KlipperScreen #189: Install script doesn't work on fresh Raspbian or Fluiddpi

charlespick, 2021-06-23:
Install script exits with this long error: 
```    
ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-q8xlxbj7/vext-gi_afa827f4c49f496bba019a9aeeba5e26/setup.py'"'"'; __file__='"'"'/tmp/pip-install-q8xlxbj7/vext-gi_afa827f4c49f496bba019a9aeeba5e26/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-uytqktd6/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-q8xlxbj7/vext-gi_afa827f4c49f496bba019a9aeeba5e26/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-uytqktd6/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-q8xlxbj7/vext-gi_afa827f4c49f496bba019a9aeeba5e26/setup.py'"'"'; __file__='"'"'/tmp/pip-install-q8xlxbj7/vext-gi_afa827f4c49f496bba019a9aeeba5e26/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-uytqktd6/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.
```

something about the install script causes apt to throw a fit: 
`sudo apt upgrade`
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 xserver-xorg-input-evdev : Depends: libevdev2 (>= 1.2.2+dfsg-1~) but it is not installed
                            Depends: libmtdev1 (>= 1.1.0) but it is not installed
                            Depends: xorg-input-abi-24
                            Depends: xserver-xorg-core (>= 2:1.18.99.901) but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

and starting screen.py in the virtualenv causes this:
```
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 4, in <module>
    import gi
ModuleNotFoundError: No module named 'gi'
```

darkuncle33, 2021-06-28:
I have the same issue, with same error message. 
After reboot, I can no longer access the web interface.
Klipper screen will display on the touch screen but will not connect to the printer.

jordanruthe, 2021-07-18:
The install script was unable to install the proper dependencies due to apt being in a broken state. Commit 145d9c1d8403cb6e19ba0098a7f190ee2c5e0092 should help with this issue as it will attempt to fix broken packages. Please update and re-run the install script.

Just as a note, you can safely ignore the errors for vext.gi

closed 2021-08-29 by jordanruthe