KlipperScreen/KlipperScreen #191: Klipperscreen does not start after install.

Alex00101, 2021-06-24:
Installing Klipperscreen on PI3 (please note that I used a standalone PI with all the software but ot connected to the actual printer if it matters). 

Encountered an error during install listed below. After searching similar issues (incomplete setup on RPI4 (with KIAUH) #26) issued  sudo systemctl enable KlipperScreen and sudo systemctl start KlipperScreen and can see the message "Stared Kliperscreen" on startup but PI boots to the console. 

May I please ask for assistance?

Thanks

Installing collected packages: vext.gi
    Running setup.py install for vext.gi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-u9ccgk9x/vext-gi_7caafaeff976430795a46dd7a4c73ef9/setup.py'"'"'; __file__='"'"'/tmp/pip-install-u9ccgk9x/vext-gi_7caafaeff976430795a46dd7a4c73ef9/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-hldniri9/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-u9ccgk9x/vext-gi_7caafaeff976430795a46dd7a4c73ef9/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-hldniri9/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-u9ccgk9x/vext-gi_7caafaeff976430795a46dd7a4c73ef9/setup.py'"'"'; __file__='"'"'/tmp/pip-install-u9ccgk9x/vext-gi_7caafaeff976430795a46dd7a4c73ef9/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-hldniri9/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.
Installing KlipperScreen unit file
Created symlink /etc/systemd/system/multi-user.target.wants/KlipperScreen.service → /etc/systemd/system/KlipperScreen.service.
Adding X11 Xwrapper
allowed_users=anybody
pi@Troodon30K:~/KlipperScreen $ sudo systemctl enable KlipperScreen
pi@Troodon30K:~/KlipperScreen $ sudo systemctl start KlipperScreen

jordanruthe, 2021-06-24:
Check for the `/tmp/KlipperScreen.log` file. Logging output should go there
and have any issues (attach it to this issue if you need help).

If that file does not exist, run `journalctl -xe -u KlipperScreen` and that
will give you the error output you can paste here for help.

Alex00101, 2021-06-24:
Thanks a lot and apologies, I'm not an advanced user. 2 log files are attached (there was no /tmp/KlipperScreen.log file) to the issue post. Hope it makes sense....

Thanks

jordanruthe, 2021-06-27:
This seems like something didn't install quite right. Can you run a couple of commands for me and put the output here?
```
ll /dev/tty[0-9]
cat /etc/group | grep tty
```

Alex00101, 2021-06-28:
Thank you. I've done a clean install (with the same result) and additional info is below:

ll /dev/tty[0-9]  (got an error that ll is not found, so below is the output from ls -al /dev/tty[0-9]
**pi@TroodonTst:~ $ ls -al /dev/tty[0-9]**
crw--w---- 1 root tty 4, 0 Jun 28 00:50 /dev/tty0
crw--w---- 1 root tty 4, 1 Jun 28 00:50 /dev/tty1
crw--w---- 1 root tty 4, 2 Jun 28 00:50 /dev/tty2
crw--w---- 1 root tty 4, 3 Jun 28 00:50 /dev/tty3
crw--w---- 1 root tty 4, 4 Jun 28 00:50 /dev/tty4
crw--w---- 1 root tty 4, 5 Jun 28 00:50 /dev/tty5
crw--w---- 1 root tty 4, 6 Jun 28 00:50 /dev/tty6
crw--w---- 1 root tty 4, 7 Jun 28 00:50 /dev/tty7
crw--w---- 1 root tty 4, 8 Jun 28 00:50 /dev/tty8
crw--w---- 1 root tty 4, 9 Jun 28 00:50 /dev/tty9

**pi@TroodonTst:~ $ cat /etc/group | grep tty**
tty:x:5:pi

**pi@TroodonTst:~ $ cat /etc/X11/Xwrapper.config**
allowed_users=anybody

In addition to that, below is an error I've received when running the KlipperScereen install.sh

ERROR: Command errored out with exit status 1:
     command: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-uhy9bfkz/vext-gi_3fece1b781784c4fb5289f51ce1a0b93/setup.py'"'"'; __file__='"'"'/tmp/pip-install-uhy9bfkz/vext-gi_3fece1b781784c4fb5289f51ce1a0b93/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-b23rfenk/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi
         cwd: /tmp/pip-install-uhy9bfkz/vext-gi_3fece1b781784c4fb5289f51ce1a0b93/
    Complete output (22 lines):
    running install
    vext.gi Install
    running build
    running install_data
    creating /home/pi/.KlipperScreen-env/share/vext
    creating /home/pi/.KlipperScreen-env/share/vext/specs
    copying gi.vext -> /home/pi/.KlipperScreen-env/share/vext/specs
    running install_egg_info
    running egg_info
    writing vext.gi.egg-info/PKG-INFO
    writing dependency_links to vext.gi.egg-info/dependency_links.txt
    writing requirements to vext.gi.egg-info/requires.txt
    writing top-level names to vext.gi.egg-info/top_level.txt
    reading manifest file 'vext.gi.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    warning: no files found matching '*.txt'
    writing manifest file 'vext.gi.egg-info/SOURCES.txt'
    Copying vext.gi.egg-info to /home/pi/.KlipperScreen-env/lib/python3.7/site-packages/vext.gi-0.7.4-py3.7.egg-info
    running install_scripts
    writing list of installed files to '/tmp/pip-record-b23rfenk/install-record.txt'
    Install vext files:
    error: [Errno 2] No such file or directory: 'vext': 'vext'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.KlipperScreen-env/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-uhy9bfkz/vext-gi_3fece1b781784c4fb5289f51ce1a0b93/setup.py'"'"'; __file__='"'"'/tmp/pip-install-uhy9bfkz/vext-gi_3fece1b781784c4fb5289f51ce1a0b93/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-b23rfenk/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.KlipperScreen-env/include/site/python3.7/vext.gi Check the logs for full command output.

I hope it helps/provides requested info.

jordanruthe, 2021-06-28:
Thanks for the info, the vext.gi error isn't relevant (it errors even on good installations).

Can you try reinstalling the xserver-xorg by running `apt purge xserver-xorg`

Once you have done that, re-run the installation script.

Alex00101, 2021-06-28:
Thanks, tried that, unfortunately the result was the same. Happy to provide more logs or troubleshoot further.

jordanruthe, 2021-06-28:
I hate asking this, but can you attempt building on a fresh image? What
image are you using to build this?

Alex00101, 2021-06-28:
Absolutely happy to try any other image. I've tried 2 so far, 1st the one with octoprint, klipper and mainsail prepared for me by an expert in my printer (so it is working except the KlipperScreen) adn second the fresh Mainsail image from mainsail if it makes sense. Results are the same. If there is a known good working image I'm happy to try that.

jordanruthe, 2021-06-29:
I generally build from raspbian-lite. I'll attempt to build from mainsailos and see if I have any issues from stock. If you have discord, feel free to jump in there (details on the contact page in the docs) as other people may have insight as well.

Alex00101, 2021-06-29:
Thanks you and apologies, if is most likely my lack if Linux knowledge. I will try raspbian lite

jordanruthe, 2021-06-29:
On mainsailos (fresh install), after running the following block of commands, KlipperScreen was displaying on my screen.

```
git clone https://github.com/jordanruthe/KlipperScreen.git
cd KlipperScreen/
./scripts/KlipperScreen-install.sh
```

Is your screen connected via HDMI or are you using a GPIO screen?

Alex00101, 2021-06-29:
Thanks, its HDMI. Let me retry the mainsailos.

jordanruthe, 2021-06-29:
I didn't mention, I modified the wpa-supplicant file to add wifi, and
config.txt to include the resolution per the hardware docs for klipperscreen

Alex00101, 2021-06-29:
Thanks, I've updated config.txt (edited my previous post with the info), my PI is on LAN, not on WiFi

closed 2021-06-29 by jordanruthe

Day-Slayer, 2022-04-21:
Having a similar issue with KlipperScreen not showing up after what seemed like a successful install. Using the BTT TFT50 V2, looks like PI starts KlipperScreen and after "Starting Permit User Sessions" the next line is "[FAILED] Failed to start /etc/rc.local Compatability". 

Possible to get some assistance with this?

Giorgiofossati, 2022-05-26:
Im having the same Issue, the screen doesnt show up klipperscreen, and i have this error: [FAILED] Failed to start /etc/rc.local Compatability.
Thanks if someone want to help us.

> Having a similar issue with KlipperScreen not showing up after what seemed like a successful install. Using the BTT TFT50 V2, looks like PI starts KlipperScreen and after "Starting Permit User [...]