KlipperScreen/KlipperScreen #207: enable touch after screen wakes up

juulius, 2021-07-20:
hi all, 

maybe somebody already figured this out but when i klick on the touchscreen to wake it up it also selects buttons on the screen. 
it now happend that i woke the screen up by touching it directly at the place where the emergency stop was wich in term ment that klipper got restarted.

could it be possible to wake up on touch and after x milliseconds enable the buttons so that it would be less lickely u would wake up the touchscreen and klick on buttons with one touch?

grigi, 2021-08-02:
I have the same issue on my BTT PiFTF50. The screen turns off, but all button presses still work. So when screen is off I have to deliberately press on the top-right as there is usually nothing too severe there.

SYNdrone-xx, 2021-08-30:
Same issue here.

Ka8oum, 2021-09-03:
yep same here, raspberry official 7" touch screen

korbnic, 2021-10-05:
+1 BTT PiFTF50v1

juulius, 2021-11-11:
Oke late night edit but had to share, got my fix working :D

so to start here is my code, (dont shoot me got almost no experience using python) 

in the screen.py in klipper i made a new def just above the main, but still part of the class KlipperScreen (note the parameter self between the brackets) 

this wil check the dpm state (everything different then the state "On" means the screen is blank")  and check if the window is visible or hidden. then will hide the window if the screen is not blank and show it when the screen is on

the counter (global so has to be defined at the top of screen.py) is there to insure that not at the first press of the screen the widow is visible making it posible to pres the button immediately

`    def check_state(self):
        'get DPMS status of screen'
        state = functions.get_DPMS_state()
        global counter

        isvisible = self.get_property("visible")
        if isvisible:
            logging.info("running check, window is visible")
        else:
            logging.info("running check, window is invisible")

        if state != functions.DPMS_State.On and isvisible:
            if counter == 1:
                logging.info("DPM state , window hidden")
                self.hide()
            else:
                counter += 1

        elif state == functions.DPMS_State.On and not isvisible:
            if counter == 1:
                logging.info("DPM state , window show")
                self.show()
            else:
                counter += 1

        else:
            counter = 0

        return True`

in the KlipperScreen init you need to add all the way down `GLib.timeout_add(1000, self.check_state)`

(full code below of the init step, not the bold text at the end witch needed to be added )
`    def __init__(self, args, version):
        self.dpms_timeout = None
        self.version = version

        configfile = os.path.normpath(os.path.expanduser(args.configfile))

        self._config = KlipperScreenConfig(configfile, self)
        self.lang = self._config.get_lang()

        self.network_interfaces = netifaces.interfaces()
        self.wireless_interfaces = [int for int in self.network_interfaces if int.startswith('w')]
        self.wifi = None
        if len(self.wireless_interfaces) > 0:
            logging.info("Found wireless interfaces: %s" % self.wireless_interfaces)
            self.wifi = WifiManager(self.wireless_interfaces[0])

        logging.debug("OS Language: %s" % os.getenv('LANG'))

        settings = Gtk.Settings.get_default()
        logging.info("Font settings: %s" % settings.get_property('gtk-font-name'))

        self.lang_ltr = True
        for lang in self.rtl_languages:
            if os.getenv('LANG').lower().startswith(lang):
                self.lang_ltr = False
                Gtk.Widget.set_default_direction(Gtk.TextDirection.RTL)
                logging.debug("Enabling RTL mode")
                break

        _ = self.lang.gettext

        Gtk.Window.__init__(self)
        self.width = self._config.get_main_config().getint("width", Gdk.Screen.get_width(Gdk.Screen.get_default()))
        self.height = self._config.get_main_config().getint("height", Gdk.Screen.get_height(Gdk.Screen.get_default()))
        self.set_default_size(self.width, self.height)
        self.set_resizable(False)
        logging.info("Screen resolution: %sx%s" % (self.width, self.height))

        self.theme = self._config.get_main_config_option('theme')
        self.gtk = KlippyGtk(self, self.width, self.height, self.theme,
                             self._config.get_main_config().getboolean("show_cursor", fallback=False),
                             self._config.get_main_config_option("font_size", "medium"))
        self.keyboard_height = self.gtk.get_keyboard_height()
        self.init_style()

        self.base_panel = BasePanel(self, "Base Panel", False)
        self.add(self.base_panel.get())
        self.show_all()
        self.base_panel.activate()

        self.printer_initializing(_("Initializing"))

        self.set_screenblanking_timeout(self._config.get_main_config_option('screen_blanking'))

        # Move mouse to 0,0
        os.system("/usr/bin/xdotool mousemove 0 0")
        # Change cursor to blank
        if self._config.get_main_config().getboolean("show_cursor", fallback=False):
            self.get_window().set_cursor(Gdk.Cursor(Gdk.CursorType.ARROW))
        else:
            self.get_window().set_cursor(Gdk.Cursor(Gdk.CursorType.BLANK_CURSOR))

        printers = self._config.get_printers()
        logging.debug("Printers: %s" % printers)
        if len(printers) == 1:
            pname = list(self._config.get_printers()[0])[0]
            self.connect_printer(pname)
        else:
            self.show_panel("printer_select", "printer_select", "Printer Select", 2)

        **GLib.timeout_add(1000, self.check_state)**`

if people got better ideas im all ears just wanted to share

alfrix, 2021-11-12:
i don't have this problem with my screens
juulius patch for testers:
 ```
diff --git a/screen.py b/screen.py
index e83b983..68b3974 100644
--- a/screen.py
+++ b/screen.py
@@ -153,6 +153,37 @@ class KlipperScreen(Gtk.Window):
         else:
             self.show_panel("printer_select", "printer_select", "Printer Select", 2)
 
+        GLib.timeout_add(1000, self.check_state)
+
+    def check_state(self):
+        state = functions.get_DPMS_state()
+        global counter
+
+        isvisible = self.get_property("visible")
+        if isvisible:
+            logging.info("running check, window is visible")
+        else:
+            logging.info("running check, window is invisible")
+
+        if state != functions.DPMS_State.On and isvisible:
+            if counter == 1:
+                logging.info("DPM state , window hidden")
+                self.hide()
+            else:
+                counter += 1
+
+        elif state == functions.DPMS_State.On and not isvisible:
+            if counter == 1:
+                logging.info("DPM state , window show")
+                self.show()
+            else:
+                counter += 1
+
+        else:
+            counter = 0
+
+        return True
+
     def connect_printer_widget(self, widget, name):
         self.connect_printer(name)
 

```

juulius, 2021-11-12:
im running a Raspberry Pi 7" DSI Touchscreen display.
dont know if the DSI port has something to do with this?

Fr33man85, 2021-11-13:
Tested and proved.
but after "touching" it takes "long" to bring up the menu. (is this depending on the 1000ms delay below?)

Had some issues when adding "GLib.timeout_add(1000, self.check_state)" to screen.py.
after removing and re-adding, it was working.. not sure, if the alignment below the "else-statement" is/was important.

want do note this.

juulius, 2021-11-13:
Because there is no event on state change when the screen is on or suspended, the code needs to pol this information actively. That's what the "GLib.timeout_add(1000, self.check_state)"  is doing. Every 1 sec the script check_state is being run.

A second thing is that in the code i wait 1 cycle before enabling the screen, this is because we don't know when the screen is pressed. It may be at the same time when the script is checking making it possible that buttons may be pressed. 

What I did to make it more responsive is to shorten the 1000 to 500 

Note that these lines where for my own logging and are clobbering the log files with unnecessary info.

if isvisible:
            logging.info("running check, window is visible")
        else:
            logging.info("running check, window is invisible")

You can just remove or comment them out

Fr33man85, 2021-11-14:
have already seen the "status information".
it was helpful for testing, but that's true. this will unnecessarily flood the journal. I commented it out

I also played around with the delay-time. 500ms seems good to me
(250ms is to quick for me, misclicks happens)

PR #340 (2021-12-04): Wakeup touch block
Should fix #207, fix #292, resolve #242
closed 2021-12-04 by jordanruthe