KlipperScreen/KlipperScreen #209: KlipperScreen Stuck on Connecting to Ender 3

whytho42, 2021-07-23:
I installed Klipperscreen and got the UI to appear on my touchscreen but it will not connect to my ender 3. I ran journalctl -xe -u KlipperScreen | cat and got this. 

● KlipperScreen.service - KlipperScreen
   Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-07-23 04:56:43 BST; 10h ago
 Main PID: 589 (xinit)
    Tasks: 22 (limit: 1939)
   CGroup: /system.slice/KlipperScreen.service
           ├─589 /usr/bin/xinit /home/pi/.KlipperScreen-env/bin/python /home/pi/KlipperScreen/screen.py
           ├─596 /usr/lib/xorg/Xorg :0
           └─684 /home/pi/.KlipperScreen-env/bin/python /home/pi/KlipperScreen/screen.py

Jul 23 15:31:07 octopi xinit[589]: DEBUG:root:Attempting to reconnect
Jul 23 15:31:07 octopi xinit[589]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
Jul 23 15:31:07 octopi xinit[589]: 2021-07-23 15:31:07,345 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Jul 23 15:31:07 octopi xinit[589]: 2021-07-23 15:31:07,347 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot
Jul 23 15:31:07 octopi xinit[589]: DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
Jul 23 15:31:07 octopi xinit[589]: 2021-07-23 15:31:07,358 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
Jul 23 15:31:07 octopi xinit[589]: DEBUG:root:Websocket error: [Errno 111] Connection refused
Jul 23 15:31:07 octopi xinit[589]: DEBUG:root:Connection already closed
Jul 23 15:31:07 octopi xinit[589]: 2021-07-23 15:31:07,364 [KlippyWebsocket.py:on_error()] - Websocket error: [Errno 111] Connection refused
Jul 23 15:31:07 octopi xinit[589]: 2021-07-23 15:31:07,365 [KlippyWebsocket.py:on_close()] - Connection already closed
...skipping...

I looked up the [Errno 111] Connection refused error and found that it is an incorrect IP address or port but I made sure the IP address is 127.0.0.1 and the port is 7125 on both moonraker.conf and KlipperScreen.conf. I do not know what is wrong any help would be greatly appreciated

jordanruthe, 2021-07-26:
Did you install on the same pi as moonraker and klipper? Can you check that the moonraker service is actually running (`systemctl status moonraker`)?

josh113n, 2021-07-27:
been having the same probablm not sure if it matters but mines on a ender3 v2 with a skr mine v2 running fluidd the screen is a btt pi tft 50 it comes up with klipper screen but all it says is connecting to printer and nothing works i did add permission in the moonraker.conf under trusted_clients: 127.0.0.1 and restarted but still coming up with this and still getting the errors 
 
if i run this  journalctl -xe -u KlipperScreen i get this

Jul 27 23:52:52 fluiddpi xinit[652]: 2021-07-27 23:52:52,687 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Jul 27 23:52:52 fluiddpi xinit[652]: 2021-07-27 23:52:52,688 [KlippyWebsocket.py:on_close()] - Connection already closed
Jul 27 23:52:53 fluiddpi xinit[652]: DEBUG:root:Attempting to reconnect
Jul 27 23:52:53 fluiddpi xinit[652]: 2021-07-27 23:52:53,159 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Jul 27 23:52:53 fluiddpi xinit[652]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
Jul 27 23:52:53 fluiddpi xinit[652]: 2021-07-27 23:52:53,160 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
Jul 27 23:52:53 fluiddpi xinit[652]: DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
Jul 27 23:52:53 fluiddpi xinit[652]: 2021-07-27 23:52:53,166 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
Jul 27 23:52:53 fluiddpi xinit[652]: DEBUG:urllib3.connectionpool:http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Jul 27 23:52:53 fluiddpi xinit[652]: 2021-07-27 23:52:53,175 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Jul 27 23:52:53 fluiddpi xinit[652]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Jul 27 23:52:53 fluiddpi xinit[652]: DEBUG:root:Connection already closed
Jul 27 23:52:53 fluiddpi xinit[652]: 2021-07-27 23:52:53,187 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Jul 27 23:52:53 fluiddpi xinit[652]: 2021-07-27 23:52:53,187 [KlippyWebsocket.py:on_close()] - Connection already closed
lines 940-1001/1001 (END)

and when i run systemctl status moonraker this is what comes back so im guessing im still missing some permissions some where 

 moonraker.service - Starts Moonraker on startup
   Loaded: loaded (/etc/systemd/system/moonraker.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-07-27 23:06:48 BST; 32min ago
 Main PID: 651 (python)
    Tasks: 4 (limit: 4915)
   CGroup: /system.slice/moonraker.service
           └─651 /home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/moonraker.py -c /home/pi/klipper_config/moonraker.conf -l /home/pi/klipper_logs/moonraker.log

Jul 27 23:39:17 fluiddpi python[651]: [web.py:log_exception()] - 401 GET /websocket?token=False (127.0.0.1): Unauthorized
Jul 27 23:39:17 fluiddpi python[651]: [app.py:log_request()] - 401 GET /websocket?token=False (127.0.0.1) [No User] 3.99ms
Jul 27 23:39:18 fluiddpi python[651]: [web.py:log_exception()] - 401 GET /access/oneshot_token (127.0.0.1): Unauthorized
Jul 27 23:39:18 fluiddpi python[651]: [app.py:log_request()] - 401 GET /access/oneshot_token (127.0.0.1) [No User] 4.08ms
Jul 27 23:39:18 fluiddpi python[651]: [web.py:log_exception()] - 401 GET /websocket?token=False (127.0.0.1): Unauthorized
Jul 27 23:39:18 fluiddpi python[651]: [app.py:log_request()] - 401 GET /websocket?token=False (127.0.0.1) [No User] 3.49ms
Jul 27 23:39:18 fluiddpi python[651]: [web.py:log_exception()] - 401 GET /access/oneshot_token (127.0.0.1): Unauthorized
Jul 27 23:39:18 fluiddpi python[651]: [app.py:log_request()] - 401 GET /access/oneshot_token (127.0.0.1) [No User] 5.11ms
Jul 27 23:39:18 fluiddpi python[651]: [web.py:log_exception()] - 401 GET /websocket?token=False (127.0.0.1): Unauthorized
Jul 27 23:39:18 fluiddpi python[651]: [app.py:log_request()] - 401 GET /websocket?token=False (127.0.0.1) [No User] 5.91ms

whytho42, 2021-08-03:
> Did you install on the same pi as moonraker and klipper? Can you check that the moonraker service is actually running (`systemctl status moonraker`)?

I ran the command and got this 
● moonraker.service - API Server for Klipper
   Loaded: loaded (/etc/systemd/system/moonraker.service; enabled; vendor preset
   Active: activating (auto-restart) (Result: exit-code) since Tue 2021-08-03 04
  Process: 2069 ExecStart=/home/pi/moonraker-env/bin/python /home/pi/moonraker/m
 Main PID: 2069 (code=exited, status=1/FAILURE)

jordanruthe, 2021-12-04:
If you're still have issues, please re-open this.

closed 2021-12-04 by jordanruthe