KlipperScreen/KlipperScreen #210: Cursor shown in print start screen

hawkeyexp, 2021-07-24:
I noticed the x-cursor is visible if you select a part with the print dialog on last step to start print...
[image]

I also checked the settings and in [main] `show_cursor: False` is set.

jordanruthe, 2021-07-26:
Is it only showing during the print dialog? Does it hide after the dialog goes away?

hawkeyexp, 2021-07-27:
After starting the print or cancel it is hidden again - i also noticed that the cursor is also shown if you are printing and cancel the print - in this screen to confirm or cancel it is also visible and window alignment is not correct like in the screenshot - on bottom ther is a small linie visible from other screen - in general all main screens like job status main menu etc. are ok and cursor is hidden.