KlipperScreen/KlipperScreen #212: Unable to successfully install KlipperScreen

hapklaar, 2021-07-25:
During installation on Pi4 with raspbian I get the following:

[image]

- I'm unable to start KlipperScreen. 
- no /tmp/KlipperScreen.log exists yet
- `sudo journalctl -xe -u KlipperScreen` gives
[image]
- /etc/X11/Xwrapper.conf has entry `allowed_users=anybody`
- when I do a `cat /etc/group | grep tty` I see my username

Any clue what might be wrong?

hapklaar, 2021-07-25:
I did get it to work, but had to do some additional configuration that was not specified in the installation instructions:

- group had to be able to read from /dev/tty2 apparently >> `sudo chmod g+r /dev/tty2'
- my account had to be member of the video group as well >> `sudo usermod -a -G video $USER`

after this I was able to start with `sudo systemctl start KlipperScreen.service`

hapklaar, 2021-07-31:
To get touchscreen to work, I had to make another permission change:
`sudo chmod o+rw /dev/input/*`

jeffreynichols, 2021-08-11:
Thanks! that helped me finish my installation.

LoganFraser, 2021-08-11:
I'm getting the same error, and while I can start it manually now with 'sudo systemctl start KlipperScreen.service' after a restart I get a login prompt and the error if I run a 'sudo journalctl -xe -u KlipperScreen' again.

closed 2021-09-01 by jordanruthe