KlipperScreen/KlipperScreen #213: i  can't  connect to Wi-Fi

jediseed, 2021-07-26:
hi ,i use fluidd os and kilpper ，when i connect 5 inch touch screen PI 3B+, the wifi can't connect ,that my log,do you konw  what happend?
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/6878887/KlipperScreen.log)

jordanruthe, 2021-07-26:
Are you trying to connect to wifi with KlipperScreen or are you saying that wifi is disconnected when using KlipperScreen?

I see that KlipperScreen is unable to identify the currently connected wifi network. Unfortunately, I'm not able to ascertain more information as I do not log wifi network names for privacy reasons.

jediseed, 2021-07-27:
I found that when I use 2.4G wireless network，Unable to connect when pi  power turned on。
I try to use 5.8G network，it can connect。I think it may be solved，need more test,  thank you

closed 2021-07-27 by jediseed