KlipperScreen/KlipperScreen #217: KlipperScreen shows estimate from a previous print

Drachenkaetzchen, 2021-07-29:
I have printed a very small calibration file (let's name it `foo.gcode`), which took roughly 5 minutes.

Afterwards, I cleared the build plate in my slicer and sliced a way bigger file. When I now started printing the bigger file (with the same `foo.gcode` name), KlipperScreen used the time data from the previous print in the job status display. The percent display went to 100% after 5 minutes (which was the previous print time) and also the remaining time was stuck at 0:00.

The thumbnail displays correctly.

All times and percentage in `fluidd` match the new file correctly.

# Steps to reproduce

These steps are untested, as my large print is currently printing.

- Create a small test print with the name `foo.gcode`
- Print a very small file using `SDCARD_PRINT_FILE FILENAME=foo.gcode`
- Replace `foo.gcode` with a significantly larger print
- Print the same file again
- The time information in the job status display show times from the previous smaller print

alfrix, 2021-09-13:
probably fixed along with #227 with [31606dd](https://github.com/jordanruthe/KlipperScreen/commit/31606ddc380ad9eb0d6aeb05913b2f6c167a7c37)

jordanruthe, 2021-09-20:
Please re-open this if you're still having issues.

closed 2021-09-20 by jordanruthe