KlipperScreen/KlipperScreen #222: Print time estimate is behaving very strange...

hapklaar, 2021-08-08:
Whenever I start a print, the time-left estimate is way off. It starts ok-ish with time-left estimate close to what's expected and what Fluidd is showing. 

But then it starts to climb very rapidly. It's not uncommon for a small print, taking around 30 minutes, to show 200+hours on the estimate after a minute of printing.

After a while this way too high estimate starts dropping and returns back to a normal value.

phil2sat, 2021-08-15:
have this behavior too, wouldnt it be possible to implement same algo as fluidd does, i mean klipperscreen has same data as fluidd has

ichbaum, 2021-09-08:
After 30min I'm still way of.
>5h on the screen and an Estimate of 3:10 on the web interface.

alfrix, 2022-03-05:
Closed as duplicated of #108

PR #378 (2022-03-10): Rework Time estimates
It should be much more accurate
Fixes #222
Fixes #108 
Fixes #282
closed 2022-03-05 by alfrix