KlipperScreen/KlipperScreen #227: File refresh on KlipperScreen does not work

hapklaar, 2021-08-13:
I upload gcode files through Fluidd in the web interface. These files only become visible on KlipperScreen after I restart the KlipperScreen process. The refresh button in KlipperScreen file browser doesn't seem to do anything.

As no one else has reported this, I assume it is an issue specific to my system. 

What could it be?

jordanruthe, 2021-08-14:
Please upload your `KlipperScreen.log` file from when you notice the issue occurring. It's located in `/tmp/`

hapklaar, 2021-08-14:
Updated all components (klipper, klipperscreen, moonraker) today and it is working fine now. Issue solved!

hapklaar, 2021-09-10:
Hi @jordanruthe. This issue still occurs when I upload a file that has the same filename as a previous print. I think the file details are not updated because it is detected as a duplicate:

```
2021-09-10 12:20:21,856 [screen.py:show_panel()] - Showing back. count: 0
2021-09-10 12:20:21,860 [screen.py:show_panel()] - Current panel hierarchy: ['main_panel']
2021-09-10 12:20:21,860 [KlippyWebsocket.py:object_subscription()] - Sending printer.objects.subscribe: {'objects': {'bed_mesh': ['profile_name', 'mesh_max', 'mesh_min', 'probed_matrix'], 'configfile': ['config'], 'display_status': ['progress', 'message'], 'fan': ['speed'], 'gcode_move': ['extrude_factor', 'gcode_position', 'homing_origin', 'speed_factor'], 'idle_timeout': ['state'], 'pause_resume': ['is_paused'], 'print_stats': ['print_duration', 'total_duration', 'filament_used', 'filename', 'state', 'message'], 'toolhead': ['homed_axes', 'estimated_print_time', 'print_time', 'position', 'extruder'], 'virtual_sdcard': ['file_position', 'is_active', 'progress'], 'webhooks': ['state', 'state_message'], 'extruder': ['target', 'temperature', 'pressure_advance', 'smooth_time'], 'heater_bed': ['target', 'temperature']}}
2021-09-10 12:20:23,153 [screen.py:_websocket_callback()] - Filelist changed: {
  "action": "create_file",
  "item": {
    "modified": 1631269223.0140607,
    "size": 1326172,
    "path": "BB1_Tube_Squeezer_Rod.gcode",
    "root": "gcodes"
  }
}
2021-09-10 12:20:23,153 [files.py:add_file()] - File already exists: BB1_Tube_Squeezer_Rod.gcode
2021-09-10 12:20:27,782 [screen.py:show_panel()] - Attaching panel main_panel_print
2021-09-10 12:20:27,817 [screen.py:show_panel()] - Showing back. count: 1
2021-09-10 12:20:27,819 [screen.py:show_panel()] - Current panel hierarchy: ['main_panel', 'main_panel_print']
2021-09-10 12:20:29,921 [KlippyWebsocket.py:get_file_list()] - Sending server.files.list
```
and results in the file list not being updated with this new version of the same file. Maybe you can do a check on the 'modified' property to see if it is the same, or just reprocess any file that is uploaded, regardless if it already exists?

commit 31606ddc (2021-09-12): Update metadata on file change. Fixes #227
closed 2021-09-12 by jordanruthe