KlipperScreen/KlipperScreen #235: Print screen alignment fails if filename to long

hawkeyexp, 2021-08-21:
Hi, if a filename is longer and an auto linebreak is done it could happen that the buttons on bottom moves out of screen and be only visible partially. In my case the filename split has used 5 lines where i noticed this behaviour.
I think a limit for linebreaks would be usefull or possible only showing the first line in big font and if a linebreak is needed the other lines in smaller fonz size.

alfrix, 2021-08-22:
A posible solution would be single line label (nowrap) and ellipsize end

alfrix, 2021-08-22:
Long filename tests, 2 lines seems to be the maximum
[image]
[image]
[image]

M117 also causes problems and should be ellipsized
[image]

hawkeyexp, 2021-08-28:
Looks good to me now

closed 2021-09-01 by jordanruthe