KlipperScreen/KlipperScreen #236: Z Calibrate panel and Voron printers

SYNdrone-xx, 2021-08-22:
Z Calibrate panel call PROBE_CALIBRATION gcode but on Voron printers should call/manage Z_ENDSTOP_CALIBRATE.
Is it possible to add an option to choose between PROBE_CALIBRATE and Z_ENDSTOP_CALIBRATE ?

Fr33man85, 2021-09-04:
Same issue here:
Voron 2.4
BTT Octopus
Raspi 4 + 7" TS

if i click on "Settings - Z - Calibrate", the printer homes and right after Z-Homing, the nozzle crashes into Z-Endstop.

Cause: The button calls "PROBE_CALIBRATION" instead of "Z_ENDSTOP_CALIBRATE"

It will not be a big deal, if the gantry will move to the bed center in front of "PROBE_CALIBRATION"
But without XY movement, there is no "probeable" material behind the hotend.

I (and even more "Hagbard" from Voron discord) found out, that the Problem is this script:
https://github.com/jordanruthe/KlipperScreen/blob/master/panels/zcalibrate.py

Row 85:
I changed from:

`   def activate(self):
        if self._screen.printer.get_stat("toolhead","homed_axes") != "xyz":
            self._screen._ws.klippy.gcode_script(KlippyGcodes.HOME)
        self._screen._ws.klippy.gcode_script(KlippyGcodes.PROBE_CALIBRATE)`

to:
`       def activate(self):
        if self._screen.printer.get_stat("toolhead","homed_axes") != "xyz":
            self._screen._ws.klippy.gcode_script(KlippyGcodes.HOME)
        self._screen._ws.klippy.gcode_script("G90")
        self._screen._ws.klippy.gcode_script("G0 X150 Y150 F3000")
        self._screen._ws.klippy.gcode_script(KlippyGcodes.PROBE_CALIBRATE)
        #self._screen._ws.klippy.gcode_script("Z_ENDSTOP_CALIBRATE")`

and restarted KlipperScreen.

If you want to change to "Z_ENDSTOP_CALIBRATE" then uncommend the last line and commend out the second last line.

greetings

ckvsoft, 2021-10-18:
Hello

I don't think this is just a Voron problem. These are probably all printers where homing is not in the center. You could say in a config where XY should go.
If I edit the script now, there is a risk of ramming the nozzle back into bed after an update.

lg Chris

PR #327 (2021-11-21): Zcal panel: Support for Z_ENDSTOP_CALIBRATE
Also now it doesn't start until the button is pressed
i have left the abort button always active in case the calibration was started in the web interface, in that case you can just press start and continue with your calibration or abort.

Should Fix  #236 and also Fix #102
[image]
[image]
closed 2021-11-21 by jordanruthe