KlipperScreen/KlipperScreen #239: [FR] chamber temp on title additional to bed temp and nozzle temp

hawkeyexp, 2021-08-24:
Can you please make it possible to show a configurable 3rd temp (in most cases chamber temp) wich is shown in title line additional to bed temp and nozzle temp?

I think it would be very usefull for many users to addd a 3rd temp with a user defined sensor.

PS: Great job with klipper screen!

Fr33man85, 2021-09-12:
I want to make same request, thx

thephilz, 2022-02-14:
+1 this is key, it's not even on the main print screen (for chamber) that is kind of... ugh.

closed 2022-03-01 by alfrix

flyespresso, 2022-03-02:
@alfrix I don't think this feature request is closed--the update added additional heaters but not additional temp--so a chamber temp wouldn't be shown unless it was configured as an 'active/powered' chamber heater. We need to be able to add either as an additional to main print screen or title bar the chamber temp (it's just temp, not a chamber heater).

hawkeyexp, 2022-03-03:
I agree - it's nice that temps are now cleaned in the overview but has nothing to do with the request to make chamber temp visible in all screens by adding to top line where bed and heater are currently available :-)

commit 77aef7cf (2022-03-01): base_panel: Refactor titlebar, add heaters if there is space

close #239