KlipperScreen/KlipperScreen #247: Adding user to moonranker stuck on "connecting to printer" API not working?

ghod90, 2021-08-30:
After adding a user to moonraker to login with Klipperscreen locks out, following the instructions page I added the api changing klipperscreen.config from 
# Moonraker API key if this is not connecting from a trusted client IP
moonraker_api_key: False
to
# Moonraker API key if this is not connecting from a trusted client IP
moonraker_api_key: ~APIKEY~
No change to Klipperscreen though still stuck on connecting to printer. 
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7075351/KlipperScreen.log)

Checked previous issues saw #204 but it appears the code used to fix that is in the install I am using.

jordanruthe, 2021-09-13:
Do you have the moonraker option `force_login` set to `True`?

pariastagard, 2021-09-13:
I have mine set to True and I have the same exact problem. I've tried all of the tips I've seen for trying to get this to work with 0 improvement to KlipperScreen and it's perpetual "Connecting to Printer" 

When I installed it I had some issue with vext.gi whatever that is. I've tried uninstalling xserver, reinstalling xserver.

My Moonraker has Trusted_Clients 127.0.0.1

I don't know why I'm having so much trouble with this. The Fluidd and Klipper install is less than two weeks old. Everything is updated. I'm running out of ideas...beyond returning the screen and sticking with the mini12864 (or whatever the number is).

ghod90, 2021-09-13:
Yes force login was set to True. My computers can login fine also so it wasn't the Pi.

jordanruthe, 2021-09-13:
The immediate fix would be to take off all trusted_clients and turn force_logins to false on moonraker. This then allows you to use the api_key option to properly authenticate. 

I'll have to figure out the best option to enable force_logins in the future, but it will require code changes in Klipperscreen.

jordanruthe, 2021-12-04:
@ghod90 The best way to do this is set the moonraker_api_key in KlipperScreen.conf and put force_logins (in moonraker) back to true.

closed 2021-12-04 by jordanruthe