KlipperScreen/KlipperScreen #249: Ubuntu 18.04 tablet - rotate display

aeremichev, 2021-08-31:
Hello!
Thanks for this great product!
I want to use my existing tablet (Chuwi vi8) on which I have installed ubuntu desktop 18.04 to manage my printer.
Because it has great hardware and already has a touchscreen display.

I went through an interesting way in installation and configuration
- klipper
- moonraker
- fluidd
- KlipperScreen

I have some nuances with KlipperScreen, when it starts it is displayed in vertical orientation.
I couldn't manage to rotate the screen using ubuntu via xorg.
As it turned out, xorg.conf is not created automatically, and it is also impossible to create it manually.

When executing the command "DISPLAY =: 0 xrandr -o right"
The display rotates, but the KlipperScreen itself does not scale, because of this, the buttons are not visible.

The only solution that worked for me was to change the width and height of the window in places in the KlipperScreen code itself.

````
self.height = self._config.get_main_config (). getint ("width", Gdk.Screen.get_width (Gdk.Screen.get_default ()))
self.width = self._config.get_main_config (). getint ("height", Gdk.Screen.get_height (Gdk.Screen.get_default ()))
````

And after that, turn the screen, then everything looks as it should.

Can someone tell me how to properly rotate the screen in ubuntu 18.04?
Or maybe you can add an option to the KlipperScreen to flip the display?

vzagranichnyy, 2021-08-31:
Обьясни пожалуйста как установить KlipperScreen на планшет с Raspbian.

alfrix, 2021-09-07:
There is no support for vertical mode or  rotation, you must ensure that your screen width > height, either by configuring the screen to be horizontal or by rotating before starting KS

aeremichev, 2021-09-07:
> There is no support for vertical mode or rotation, you must ensure that your screen width > height, either by configuring the screen to be horizontal or by rotating before starting KS

I have not found how to rotate the screen in xorg ubuntu 18.04 so that it is rotated before starting KS
Considering that in ubuntu 18.04 the xorg.conf file is not automatically created
Do you know how to rotate the screen?

alfrix, 2021-09-07:
i have not used ubuntu in a while, check https://wiki.ubuntu.com/X/Config
it suggests that you may find what you need in /usr/share/X11/xorg.conf.d
or create a new xorg.conf manually as described in the wiki or with `sudo dpkg-reconfigure -phigh xserver-xorg`

aeremichev, 2021-09-08:
@alfrix 
Thank you very much for your help.
Now I have succeeded.
Although I have tried this before, but this time I was more fortunate.

The `sudo dpkg-reconfigure -phigh xserver-xorg` command unfortunately does not work and this error has been around for a long time.

I managed to solve my problem through the file` /usr/share/X11/xorg.conf.d/55-monitor.conf`
file name can be anything

File contents
```
Section "Monitor"
    Identifier "DSI-1"
    Modeline "1280x800_60.00" 83.50 1280 1352 1480 1680 800 803 809 831 -hsync +vsync
    Option "PreferredMode" "800x1280_60.00"
    Option "Rotate" "Right"
EndSection
```

**Identifier** can be obtained from command `DISPLAY =: 0 xrandr`
example
```
ubuntu@ubuntu-i86:~$ DISPLAY =: 0 xrandr -q
Screen 0: minimum 320 x 200, current 1280 x 800, maximum 16384 x 16384
DSI-1 connected primary 1280x800 + 0 + 0 right (normal left inverted right x axis y axis) 0mm x 0mm
   800x1280 60.09 * +
   1280x800 59.99 59.97 59.81 59.91
```
**Modeline** from command `cvt 1280 800 60`
Where do you specify screen options
Further, in Option, the screen parameters, resolution and the fact that it needs to be rotated are registered.

closed 2021-09-09 by aeremichev