KlipperScreen/KlipperScreen #254: missing logo/symbol in different screens

Fr33man85, 2021-09-05:
Hi,

there are missing symbols in some screens. 
It shows a small box with "21 91" instead. 

i think, this should be some arrows. I can provide some pictures later.

greetings
Manuel

Fr33man85, 2021-09-08:
Hi, 
yes, sry, i forgott to send some. I do today after work.

alfrix, 2021-09-09:
interesting, for some reason your font doesn't have the glyphs "↑" "↓"
a possible fix would be to draw svg images with arrow up and arrow down (these already exist in the themes)

Fr33man85 does the "→" show up while heating the extrude or bed?
as in "50 → 210ºC"

Fr33man85, 2021-09-09:
I dont have a screen, where are horizontal arrows. Not on screen "Preheat", not on the print screen:

[image]

Fr33man85, 2021-09-12:
Hey Alfrix,
now i know what you mean and yes, same issue with the "->" arrows

jordanruthe, 2021-09-21:
@Fr33man85 Can you check to see if this update fixes it? The package is `fonts-freefont-ttf`

Fr33man85, 2021-09-22:
@jordanruthe solved! thanks

closed 2021-09-22 by Fr33man85