KlipperScreen/KlipperScreen #256: Job status panel doesn't show up if print is started from Octoprint

dodgeboy, 2021-09-07:
My job status panel shows up fine if I start the job from the KlipperScreen.  But if I start the job from OctoPrint, the KlipperScreen just stays at the main menu for the entire print.  I've searched everywhere and I can't find a config or gcode trigger that I am missing that would tell KlipperScreen a print has started.

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7121791/KlipperScreen.log)

alfrix, 2021-09-07:
Octoprint it's not native, it send commands to klipper, for that reason use Octoscreen or Octodash or something that connects to Octoprint instead

dodgeboy, 2021-09-07:
> Octoprint it's not native, it send commands to klipper, for that reason use Octoscreen or Octodash or something that connects to Octoprint instead

Thank you for your reply.  Does that mean there is no way to trigger klipperscreen to show the job status if the job starts with Octoprint?  I would really prefer to run klipperscreen, because of how tightly it integrates with klipper.  The screen updates basically instantaneously.

alfrix, 2021-09-07:
> because of how tightly it integrates with klipper. The screen updates basically instantaneously.

if you want speed and integration, use Mainsail or Fluidd along with KlipperScreen

dodgeboy, 2021-09-08:
I definitely will once it supports the spaghetti detective.

closed 2021-12-04 by jordanruthe