KlipperScreen/KlipperScreen #262: Layout issues when there are multiple extruders

alfrix, 2021-09-11:
When not heating, it's fine
[image]

But the label gets too long quickly 
[image]
[image]
[image]
[image]

long temp sensor names can cause this too
[image]

basically 3 columns is just too much for this resolution (480x320)

potential fixes: change the temp label  from 
100 -> 230 C 
to:
230 C
^100^

Tried:
[image]
[image]

2 rows can work
[image]

for long label ellipsize (set max_chars if > 4 items and resolution is low)

jordanruthe, 2021-09-11:
I do want to convert the front page to more of a table setup like fluidd or mainsail. I'll try to get to this in a few days

jordanruthe, 2021-12-04:
PR #287 will fix this when it's merged in a few days.

closed 2021-12-04 by jordanruthe