KlipperScreen/KlipperScreen #272: No working panel limits

ShohninDmitriy, 2021-09-17:
No working panel limits
[image]
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/7185025/KlipperScreen.log)

hawkeyexp, 2021-09-17:
Check your square_corner_velocity in section [printer] - i had the same issue and the logs told me that the panel is awaiting an integer but my value was 5.0 which is a float - change 5.0 to 5 for example - if it's not the reason check the log in /tmp/KlipperScreen.log for addditional errors.

ShohninDmitriy, 2021-09-17:
> 
> 
> Check your square_corner_velocity in section [printer] - i had the same issue and the logs told me that the panel is awaiting an integer but my value was 5.0 which is a float - change 5.0 to 5 for example - if it's not the reason check the log in /tmp/KlipperScreen.log for addditional errors.

Thank you, everything worked.

closed 2021-09-17 by ShohninDmitriy