KlipperScreen/KlipperScreen #275: Bed_level: screws tilt adjust doesn't update the values after running the tool

alfrix, 2021-09-17:
[image]

alfrix, 2021-09-17:
the problem was introduced with 54c47554d2c8ed1de086bda00fdd209cac28cdd2
if i revert, it works
[image]

Pr20100, 2021-09-18:
Hello, same here, the values are not updated if I run the tool on the screen for a second run (but correct in the console).
Runing v0.1.6-102-gd7b8f4f

@alfrix : What do you understand by "revert"? (order of probed points?)

jordanruthe, 2021-09-18:
@alfrix @Pr20100 I believe I fixed this, can you please confirm that it's fixed?

closed 2021-09-18 by jordanruthe

Pr20100, 2021-09-19:
@jordanruthe Yes it's fixed, thanks a lot!

commit 5d6f80a1 (2021-09-18): bed_level: Fix #275. Make the screws button disabled while command is running.