KlipperScreen/KlipperScreen #281: KlipperScreen.conf missing - Has configuration management changed? not documented

WSLAB3D, 2021-09-19:
Recently fluidd reported KlipperScreen as dirty, so I let it clean it up. Afterwards, my custom configuration for multiple printers is now gone.

I have tried to recreate the config, however now it does not exist in any of the dirs:

    ${HOME}/KlipperScreen.conf
    ${KlipperScreen_Directory}/KlipperScreen.conf
    ${HOME}/klipper_config/KlipperScreen.conf
    ${HOME}/{KlipperScreen_Directory}/ks_includes/KlipperScreen.conf

there is even no KlipperScreen.conf in the github any more.

Has configuration management changed? there is nothing updated in the documentation?

jordanruthe, 2021-09-19:
Configuration management from a user perspective has not changed. However, I did rename `ks_includes/KlipperScreen.conf` to `ks_includes/defaults.conf` in order to hopefully reduce confusion. It seems because the filename was the same, most people opted to make changes to that configuration file rather than creating a new file for custom configuration.

WSLAB3D, 2021-09-19:
Thanks, yes I see that now.

I have created a new KlipperScreen.conf in the klipper_config folder, I have included just my preheat settings and printer configs, and it is working with pulling the rest of the config from the defaults.conf.

It seems though that if I want to define one preheat setting (PETG) that I also have to define the ABS and PLA, as they do not carry forward from the defaults.conf.

I suspect that if I wanted to add a single menu item i'd have to include the whole thing? it would be nice if this could work more like how standard klipper configs treat includes, so that you only need to add customisations to your own config file, and this would allow us to take advantage of new features as they may be added in updates.

jordanruthe, 2021-09-20:
I agree with you, I'm planning on re-doing how the configuration overwrite works in the near future, but I haven't gotten to it yet. You can consider this an issue to track that and leave this open until then.

alfrix, 2022-03-05:
> I suspect that if I wanted to add a single menu item i'd have to include the whole thing?

Closed this in favor of a more descritptive #530

closed 2022-03-05 by alfrix