KlipperScreen/KlipperScreen #286: klipper screen stuck on connecting to printer, getting error 401 in log

samwells2000, 2021-09-25:
just tried installing klipper screen using a pi4 and hyper pixel 4, 
the output comes up but gets stuck on the connecting to printer screen, 

log files to follow. 

`pi@fluiddpi:~/kiauh $ journalctl -xe -u KlipperScreen | cat
-- Logs begin at Sat 2021-09-25 15:02:45 BST, end at Sat 2021-09-25 15:15:23 BST. --
Sep 25 15:14:41 fluiddpi xinit[2885]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:41 fluiddpi xinit[2885]: DEBUG:root:Connection already closed
Sep 25 15:14:41 fluiddpi xinit[2885]: 2021-09-25 15:14:41,958 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:41 fluiddpi xinit[2885]: 2021-09-25 15:14:41,959 [KlippyWebsocket.py:on_close()] - Connection already closed
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Attempting to reconnect
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,426 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,427 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,435 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,445 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Connection already closed
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,459 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,460 [KlippyWebsocket.py:on_close()] - Connection already closed
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Attempting to reconnect
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,927 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,927 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,935 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,946 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:42 fluiddpi xinit[2885]: DEBUG:root:Connection already closed
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,958 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:42 fluiddpi xinit[2885]: 2021-09-25 15:14:42,959 [KlippyWebsocket.py:on_close()] - Connection already closed
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Attempting to reconnect
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,428 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,428 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,436 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,447 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Connection already closed
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,461 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,461 [KlippyWebsocket.py:on_close()] - Connection already closed
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Attempting to reconnect
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,929 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,929 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,938 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,952 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:43 fluiddpi xinit[2885]: DEBUG:root:Connection already closed
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,966 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:43 fluiddpi xinit[2885]: 2021-09-25 15:14:43,967 [KlippyWebsocket.py:on_close()] - Connection already closed
Sep 25 15:14:44 fluiddpi xinit[2885]: DEBUG:root:Attempting to reconnect
Sep 25 15:14:44 fluiddpi xinit[2885]: 2021-09-25 15:14:44,430 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Sep 25 15:14:44 fluiddpi xinit[2885]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:44 fluiddpi xinit[2885]: 2021-09-25 15:14:44,430 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
Sep 25 15:14:44 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:44 fluiddpi xinit[2885]: 2021-09-25 15:14:44,438 [connectionpool.py:_new_conn()] - Starting new HTTP connection (1): 127.0.0.1:7125
Sep 25 15:14:44 fluiddpi xinit[2885]: DEBUG:urllib3.connectionpool:http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:44 fluiddpi xinit[2885]: 2021-09-25 15:14:44,449 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:14:44 fluiddpi xinit[2885]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Sep 25 15:14:44 fluiddpi xinit[2885]: DEBUG:root:Connection already closed
Sep 25 15:14:44 fluiddpi xinit[2885]: 2021-09-25 15:14:44,461 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
[... 912 lines cut ...]
Sep 25 15:15:22 fluiddpi xinit[2885]: 2021-09-25 15:15:22,523 [KlippyWebsocket.py:on_close()] - Connection already closed
Sep 25 15:15:22 fluiddpi xinit[2885]: DEBUG:root:Attempting to reconnect
Sep 25 15:15:22 fluiddpi xinit[2885]: 2021-09-25 15:15:22,990 [KlippyWebsocket.py:reconnect()] - Attempting to reconnect
Sep 25 15:15:22 fluiddpi xinit[2885]: DEBUG:root:Sending request to http://127.0.0.1:7125/access/oneshot_token
[... cut ...]
Sep 25 15:15:23 fluiddpi xinit[2885]: 2021-09-25 15:15:23,510 [connectionpool.py:_make_request()] - http://127.0.0.1:7125 "GET /access/oneshot_token HTTP/1.1" 401 580
Sep 25 15:15:23 fluiddpi xinit[2885]: DEBUG:root:Websocket error: Handshake status 401 Unauthorized
Sep 25 15:15:23 fluiddpi xinit[2885]: DEBUG:root:Connection already closed
Sep 25 15:15:23 fluiddpi xinit[2885]: 2021-09-25 15:15:23,525 [KlippyWebsocket.py:on_error()] - Websocket error: Handshake status 401 Unauthorized
Sep 25 15:15:23 fluiddpi xinit[2885]: 2021-09-25 15:15:23,525 [KlippyWebsocket.py:on_close()] - Connection already closed`

samwells2000, 2021-09-25:
also when 127.0.0.1 is unput into the moonraker.conf file moonraker wont enable the webui

jordanruthe, 2021-09-27:
Can you post your moonraker config?

deadmoo83, 2021-10-09:
I have the exact same issue.

Here is my moonraker.conf

`[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False
config_path: ~/klipper_config
temperature_store_size: 600
gcode_store_size: 1000
log_path: ~/klipper_logs

[authorization]
force_logins: True
cors_domains:
  *.local
  *.lan
  *://app.fluidd.xyz

trusted_clients:
  127.0.0.1
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  FE80::/10
  ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker's update manager
[update_manager]
enable_auto_refresh: True

# this enabled fluidd updates
[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd

[update_manager client KlipperScreen]
type: git_repo
path: ~/KlipperScreen
origin: https://github.com/jordanruthe/KlipperScreen.git
env: ~/.KlipperScreen-env/bin/python
requirements: scripts/KlipperScreen-requirements.txt
install_script: scripts/KlipperScreen-install.sh`

jordanruthe, 2021-12-04:
You can leave force_logins to True, but you have to use the moonraker_api_key option in KlipperScreen.conf

closed 2021-12-04 by jordanruthe